Koromix/tytools #2: Missing property 'PortNum' for USB device

eshkrab, 2015-10-12:
Trying to run tyc (or TyQT) on my mac running El Capitan (probably root of the problem), I get a `Missing property 'PortNum' for USB device` error both on `tyc list` and in TyQT, when plugging Teensies in. They show up fine on the list of /dev/tty ports. Any ideas?

Koromix, 2015-10-13:
I was not sure about updating to El Capitan but this makes the decision easier. I'll install it in the evening to investigate.

Thanks for the report, I'll get back to you as soon as I know if this is caused by El Capitan or an unrelated problem.

Koromix, 2015-10-13:
I can confirm this is due to I/O Kit changes in El Capitan, but I think I know how to fix it. I should have a working release out tomorrow evening.

Koromix, 2015-10-13:
Looks a little more complicated than I thought, because Apple changed a lot of stuff in the Darwin USB stack (https://developer.apple.com/library/mac/releasenotes/Darwin/RN_USB/Articles/10_11Changes.html).

I know exactly how to make it work, but I want to make a clean fix and test it so it'll take me a few days.

Koromix, 2015-10-19:
This bug should be fixed now, you can get a working development build at https://www.dropbox.com/s/jharipxe07cumt7/TyQt-0.6.4-4-gce5de81-osx.dmg?dl=0

Tell me if it works for you too.

eshkrab, 2015-10-21:
Works perfectly, thank you so much! Any way I could get files for the command line version?

Koromix, 2015-10-21:
It's in the bundle, you can launch it with /Applications/TyQt.app/Contents/MacOS/tyc (assuming you put the bundle in /Applications, of course).

This binary depends on /Applications/TyQt.app/Contents/Frameworks/libty.dylib but it can be built statically if it's a problem. To be honest, I'm not sure why I don't do that by default... If I ever had a good reason to make it that way, I don't remember it. I'll probably switch to static binaries for the next release.

eshkrab, 2015-10-21:
Oh, thanks! You’re saving me hours at work at a really crucial time!

Koromix, 2015-10-21:
You're welcome :)

If you are using it to address specific Teensies in a multi-Teensy setup, please not that the README visible on the GitHub project pages applies to the master branch, which will become 0.7.0.

On 0.6.x you need to use a slightly different syntax for the board tag. For example to upload to the board with serial number 128750 you have to use 'tyc upload --board "#128750" <firmware>".

Closing the issue, you can mail me or post on the PJRC topic if you have questions.

closed 2015-10-21 by Koromix