Koromix/tytools #3: Browsing to hex file

LAtimes2, 2016-01-31:
Couple of improvements:
1. Add a recently used pulldown to the hex file list in the settings window
2. If a hex file is already selected, default to that directory when opening the Browse window

Koromix, 2016-01-31:
Both requests are reasonsable and easy enough to implement, I will add this next week.

(2) makes me cringe a little because I hate softwares that don't open to the current file directory (VLC used to do that for a long time)... and yet it never occured to me to make it work in TyQt.

Koromix, 2016-02-15:
Sorry for the delay, I'm rather busy. Well, except today it seems :) The master branch should contain everything you asked for.

The "recent firmwares" list is specific to each board, I don't know if that's how you wanted it? Anyway, it is accessible under the "Upload" toolbar button/menu and in the Settings page.

You can find builds for Windows and OSX:
- Win32: https://www.dropbox.com/s/yor5p4nxuuhhzyf/TyQt-0.7.0-104-g2628768-win32.zip?dl=0
- OSX: https://www.dropbox.com/s/a59bbq9hosnrqod/TyQt-0.7.0-104-g2628768-osx.dmg?dl=0

There are many other changes, especially to the 'low-level' device code... There are regressions and warnings which you can safely ignore.

LAtimes2, 2016-02-16:
Both look good. Thanks. tyqt is great :)

closed 2016-02-16 by LAtimes2