Koromix/tytools #4: Suggestion for improvement

universpharmacie, 2016-02-11:
Not an issue, but a suggestion for an improvement : Put the board names as defined in ty/board_teensy.c in the ty_board_model.name struct into an external (ini, cnf or plist or whatsoever) file which is read at runtime, so that on might give custom names to the one or the other board without re-compiling the TyQt app.

Theremingenieur, 2017-05-17:
May I ask ?

Koromix, 2017-05-21:
Yes, you may.

Obviously I completely ignored this and I'm sorry for that. I made some of the necessary changes some time ago but it was never finished.

I've pushed a branch called **patch-models-ini** that supports this. It looks for a file called **TyTools.ini** in the same directory as the executable. For now, you can only rename boards.

For exemple you can rename the Teensy LC with:
```
[Models]
Teensy = Test
Teensy LC = Test LC
```

The *Teensy* model is the generic name until the precise model can be identified, and it is used to make the board identifier (e.g. *<serial>-Teensy*). You probably don't want to put spaces in it.

So far I've only tested the branch on Windows.

Koromix, 2017-05-21:
You can find builds for this branch on BinTray: https://bintray.com/koromix/tytools/tytools/patch-models-ini/view/files

Koromix, 2017-05-26:
The latest build in this branch now:
- Parses TyTools.ini from `C:/Users/USER_NAME/AppData/Roaming/TyTools/` (and `~/.config/TyTools/` on Linux, `~/Library/Preferences/TyTools/` on OSX in addition to the executable directory
- Supports changing the supported VID:PID list.

An example for the latter:

```
[Devices]
0123:abcd = Teensy
abcd:0123/hid = Teensy
```

With this in TyTools.ini, libty will try to open devices with 0x0123:0xABCD and HID devices with 0xABCD:0x0123 VID:PID pair as Teensies.

Koromix, 2017-05-28:
The new release (0.8.5) does what you needed, though it needs documentation. You can use the posts above to use this feature.

I'm closing this issue, but if you find a bug or need something more you can still comment here.

closed 2017-05-28 by Koromix