Koromix/tytools #7: Not more than 18 Teensies displayed

jbliesener, 2016-11-03:
Dear Koromix,

I'm working with Robert Archer on his 737 Flight simulator, which uses 27(!) Teensies. The problem is that only 18 show up in `tyc list` or `tyqc`, while the log file shows they're all scanned.

Do you have any idea on that?

I thought it might have to do with the number 64 that appears as hash table (htable) size. Tried to increase that to 256, but no change.

Probably has something to do with your hash table implementation, right?

[TeensyLog.txt](https://github.com/Koromix/ty/files/569942/TeensyLog.txt)
[tyclistout.txt](https://github.com/Koromix/ty/files/569945/tyclistout.txt)

Koromix, 2016-11-04:
Hi,

From what I can see, the invisibles Teensies are plugged in a an Etron USB 3.0 host controller. TyQt and tyc rely on the USB topology to tell devices apart, and third-party host controller drives often expose broken topology information, if anything.

Could you try running TyQt in XP compatibility mode to see if it improves anything? There's an alternate code path for resolving the topology on XP. You can set the XP compat mode in the executable or shortcut properties.

jbliesener, 2016-11-05:
I tried, couldn't make it work. We're running on a Windows 7 Home Premium PC, no XP compatibility mode. Tried to install Microsoft Virtual PC, didn't work either. TyQT fails to detect anything and shows "Unsupported root HUB at 'VMUSB\HUB\6&1DDB10CA&0'.

Also tried to install latest version of Etron USB driver (v. 0.119), no change.

Could you provide a command line parameter to force XP topology detection?

jbliesener, 2016-11-05:
Just another update: I downloaded and installed the 32 bit version on 64 bit Windows 7 Home Premium. Same result, only 18 Teensies found

Koromix, 2016-11-05:
I've pushed a few commits to fix TyQt under Virtual PC (at least in theory) and to add an environment variable to force the XP code path.

You can find a build on my BinTray page at [https://bintray.com/koromix/ty/ty#files](https://bintray.com/koromix/ty/ty#files), it is named "TyQt-0.7.5-214-g7b4eb95".

To run tyc in this mode:
`set LIBHS_WIN32_FORCE_XP_LOCATION_CODE=1
tyc list`

Note that you can also set TY_DEBUG to 1 to show the full log (same as TyQt) when running tyc.

jbliesener, 2016-11-05:
That worked. Seeing 27 Teensies now.

jbliesener, 2016-11-05:
I put LIBHS_WIN32_FORCE_XP_LOCATION_CODE into system environment variables configuration and now it starts up correctly, running the 32 bit version of Ty on a 64 bit Win7. It's a workaround solution, but it works for now. Maybe you could provide this as a configuration entry on the preferences page. 

I understand that this works with the 32 bit version only, right?

Koromix, 2016-11-05:
It works for both versions, I use Win32 for all Windows-related code because most of that code sits on top of the Win32 subsystem. The XP code path is heavier and can generate USB traffic during enumeration, so I don't want to make it the default method of enumeration.

The latest build, TyQt-0.7.5-219-g420469e, should now work without LIBHS_WIN32_FORCE_XP_LOCATION_CODE. It tries the Vista code path first and the XP code path only for devices where the first method fails. You need to remove LIBHS_WIN32_FORCE_XP_LOCATION_CODE to make sure it works correctly.

I'll mark this bug as solved if you can report to me that TyQt / tyc work correctly without LIBHS_WIN32_FORCE_XP_LOCATION_CODE.

jbliesener, 2016-11-05:
I was about to install -219-g420469e when I saw that there's a -220 already. Which one do you want me to install?

Koromix, 2016-11-05:
Take 220, it's pretty much the same. I just pushed a fix for a stupid bug that would only happen in debug builds running on a real Windows XP. And new builds are generated whenever I push.

jbliesener, 2016-11-05:
-220 seems to work without the environment variable. All 27 Teensies found in tyc and tyqt.

Thank you so much!

Koromix, 2016-11-05:
Thank you, I knew there were problems with some USB 3.0 controllers before could not do much about it. I'm happy to see this now works better.

closed 2016-11-05 by Koromix