Koromix/tytools #9: Serial font still proportional on windows

plgDavid, 2017-02-20:
Hello, thanks for the great app such a blast to use for a Teensy poweruser.
My only (big) complaint is the Serial font which is still proportional as of TyTools-0.8.0-102-g7b89d86-win64. 
Maybe you got this issue?
http://stackoverflow.com/questions/18896933/qt-qfont-selection-of-a-monospace-font-doesnt-work
Thanks!

plgDavid, 2017-02-20:
[image]

plgDavid, 2017-02-20:
Forgot to mention. Im on Win7x64

Defragster, 2017-02-22:
I see this on build-102 on Win 10x64 as well.
not sure if possible - but being able to select Fixed or Variable spacing would be ideal. Sometime fixed is best to view 'table' like data - other times large amounts of text are better shown 'compressed' with the variable font.
[image]

plgDavid, 2017-02-22:
Hi
Sure the absolute best would be a font selector box which is recalled in prefs.
Cheers

Koromix, 2017-02-23:
Tonight I'll fix the broken monospace change (works on Linux, didn't bother to test it further), and add a font selector during the weekend.

It'll probably a global setting, since there is already a bunch of per-board settings and I'd like to avoid bloat there. Unless you think it would be much better per-board.

plgDavid, 2017-02-23:
Hello Koromix! Again thanks for this incredible app, you can't believe to know how good it is NOT to have to use the Arduino window ever again (was only using it for Serial USB MIDI I/O). 
Now being ALSO able to program/reset various devices and have it all working is priceless!

Back on topic. I don't see why the user selected font should be per board myself. \
More of a global fire and forget pref IMHO, Cheers

Defragster, 2017-02-23:
Global Okay, nice if per instance. One window might be one way and Ctrl+New window the other

Koromix, 2017-02-23:
I've pushed a fix, it's now monospace on Windows too. I'll try add to get font selection in during the weekend.

Defragster, 2017-02-23:
Indeed TyTools-0.8.0-107-ge5c1e9a-win64.zip is REALLY WIDE MONOSPACE.

plgDavid, 2017-02-24:
TyTools-0.8.0-107-ge5c1e9a-win64 is perfect for me! thank you!