Koromix/tytools #10: Support for 'generic Serial' devices

Defragster, 2017-02-22:
This is working well - with minor issues:

- The device I'm using goes offline for some time and is dropped some 1 to 5 seconds before it comes back online so prior text is lost and I need to reselect and serial enable, Would be nice to have a settable timeout.

- FTDI adapter is found by the IDE, but not TyCommander

- Alternate devices report a unique serial# - not in the Teensy style, but it would be great if the serial # was treated as unique for 'Options' storage, and logging of Serial data to a file.

- Would be handy if 'Generic' device would denote the COM# on the device info tab as that ties to the IDE port identification.

jbliesener, 2017-02-23:
Defragster,

I ran into the same problem and developed a partial fix for it. Tried to create a Pull Request, but didn't work. The issue is in line 820 of monitor_win32.c. Changing

`if (strncmp(dev->key, "USB\\", 4) == 0) {`

to 

`if (strncmp(dev->key, "USB\\", 4) == 0 || strncmp(dev->key,"FTDIBUS\\", 8) ==0) {`

detects the FTDI chips, but still does not work with physical COM ports. But maybe it helps you for the moment

Koromix, 2017-02-23:
Gladto see it's enough to make it work. I knew about the FTDIBUS prefix thing from reading libusb and pyserial, but since I don't have any FTDI serial device (yet) I didn't want to commit untested stuff.

But if it works at least a bit for you it's probably a good start until I can test it myself. I'll push the change tonight.

jbliesener, 2017-02-23:
Next question is how it behaves with Prolific chips. I have one, but it's an old version and I can't find any driver for Windows 10. If there's any news, I'll post it here.

jbliesener, 2017-02-23:
Oh, and Niels, please update the libhs repository as well with this modification. From what I understand, the libhs project is just an extracted and processed version of the libhs directory in tyqt. Great stuff!

Koromix, 2017-02-23:
I will. Though the libhs repository does not exist anymore, because it was annoying to maintain a separate but not-quite separate repository for the code, and it was frequently outdated and out-of-sync.

However, I've simplified libhs and I now distribute it as a single-file library, which gets built from from the multiple libhs files. That makes it as easy to use as C/C++ toolchains allow.

More information on the dedicated page: http://neodd.com/libraries/

jbliesener, 2017-02-23:
Yes, the single file distribution is what I'm using. Very nice indeed.

Koromix, 2017-02-23:
By the way, this amalgamated file is produced automatically when you build tytools. Pick "libhs.h" in the build directory.

Koromix, 2017-02-23:
I have committed the change proposed by jbliesener for FTDI devices, it is in the latest build. I don't have any FTDI device yet so can't confirm it works well.

I may add the unique flag to generic devices during the weekend. The rest will come a bit later.

Defragster, 2017-02-23:
My SparkFun FTDI is recognized!   Using: TyTools-0.8.0-107-ge5c1e9a-win64.zip 

Problem - it needs to be told the baud rate.  When I open the FTDI in IDE it has a box to tell is what baud rate to use like 115200 or 230400 - very limited choices in the IDE SerMon - that is the MAX offered.

Once I set the rate in SerMon I can connect with TyComm and that baud rate is still set and works!

This will have to be added for complete support.  Seems the FTDI could use this fuller range: { 1200, 2400, 4800, 9600, 19200, 38400, 57600, 115200, 230400, 460800, 921600 }.  Those are the directly supported speeds my 'other' device might connect, it seems FTDI can also hit 1.2, 1.5, 2.0 and 3.0 Mbaud. From: http://www.ftdichip.com/Support/Knowledgebase/index.html?whatbaudratesareachieveabl.htm

Koromix, 2017-05-06:
The latest build (TyTools-0.8.0-143-gba1b864) should configure all serial devices to 115200 bauds. Eventually I'll make it configurable but I hope it helps in the meantime?

Defragster, 2017-05-06:
Yes, Having it set to a known is good start.

Defragster, 2017-05-06:
Any chance of fixed or adjustable way to extend 'device offline timeout period' before it is dropped? { per OP }

The STM32's need at least 5-10 more seconds delay ( maybe 20 with larger code I haven't gotten to yet ).

The DFU protocol used takes device offline after 'Serial disable in TyComm' - then the programming time is long - currently device goes away - is deselected.  When the timing works TyComm Serial can be enabled as soon as DFU programming starts - and if programming completes before being dropped then TyComm picks it up immediately with no loss of startup data, and no racing around with the MOUSE is needed waiting for the upload to complete.

Koromix, 2017-05-06:
I think detecting when DFU is running to avoid dropping the board would be a better solution.

To help you until I do this, I have added a quick-and-dirty workaround in TyTools-0.8.0-145-g9258ac6: you can change the drop delay by setting the environment variable **TYTOOLS_DROP_BOARD_DELAY** to the desired delay (in milliseconds). So _TYTOOLS_DROP_BOARD_DELAY=30000_ will get you 30 seconds.

You need to restart TyQt when you change this variable.

Defragster, 2017-05-06:
Got ver 0.8.0-146-xxx and setting global ENV var in system worked to get 30 Seconds - Thanks!  That fixes item #1 in OP!!! 

I'm not sure how waiting 'too long' to drop would be a problem - having it not drop too soon will make program uploads run better. I just set it to 60 seconds - works perfectly.

Having DFU detection would be nice - from IDE or STINO_SublimeText the device is detected by connection COM##. Having a command to TyComm to auto drop and wait for reconnect would be ideal - but I don't see a way to handle that - would take a way to 'tell' TyComm when and what device.

Defragster, 2017-05-06:
The OP 4th item show device tab as:: 'Generic: COM6' would be handy to quickly ID multiple 'non-teensy' devices since SER# doesn't allow naming. As it is I plug in and hit 'Information' tab to tell them apart -  sometime multiple times to get the right device.

Koromix, 2017-05-08:
In TyTools-0.8.0-147-gbd14087, settings and serial log are enabled for generic boards with a serial number that looks unique (right now that means not full of 0s, underscores and spaces).

Defragster, 2017-05-08:
Good to see. The loss of the log from other issue was a regression not just from Ser# issue above

Defragster, 2017-05-08:
My STM32's serial # are recognized as unique { I can name the devices and see recorded log info } TyComm is a very useful and helpful tool - Thanks.

I suppose this issue could almost be closed as the OP list of issues is usable now - with a feature/Issue note for FTDI refinement and perhaps the custom DFU detection enhancements?

Koromix, 2017-05-14:
I put TODO notices in the code about that, which I review regularly since my IDE shows them in a dedicated pane.

As a bonus, any application can ask TyCommander to detach or attach the serial monitor with `TyCommanderC.exe attach --board='<identifier>'` and `TyCommander detach --board='<identifier>'` (the syntax for identifier is explained in the "Using tycmd" section of the README). This change is not persisted unless you pass `--persist` too.

You can also detach (or attach) all boards with `TyCommander detach --multi`. There are other possibilities, I need to document this someday.

closed 2017-05-14 by Koromix

jbliesener, 2017-05-14:
I'd just like to make a comment about the "original" issue of this post: generic serial devices. Those DB-9 or DB-25 ports directly on the motherboard. My comment from Feb 23rd partly resolved it for FTDI USB devices, but still left the original 16550 UART chips as a TO DO. 

Looking at the source code, I didn't find a patch that would make it work (may be my fault, I don't have a generic com port in windows to test it). Some time ago, the discussion of this issue moved towards baud rate configuration and detection timeout and maybe the original issue wasn't resolved. Or was it and I missed it?

Defragster, 2017-05-14:
When I created should have been 'generic **USB** Serial'.  Hardware serial is so rare - more likely through USB adapter so that would be different.

Thanks for the TODO notes.  Will give TyCommanderC a try if I can extract the DFU device ser #'s and put them in build process.