Koromix/tytools #12: TYQT Crashes on start

xorionfe, 2017-02-23:
There seems to be a conflict with my MSI Motherboard onboard USB drivers.   It uses Asmedia drivers.   I had the same issue on an ASUS MB that was using Asmedia USB drivers.   I also see the problem if VIA USB drivers are present.  If I disable these then TYQT will start and work but I lose many of my usb ports in the process.   Can you please look into what might be causing the problem.   I am using the latest version 8 of TYQT.

Here is the error in the crash:
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	TyQt.exe
  Application Version:	0.0.0.0
  Application Timestamp:	581e0930
  Fault Module Name:	TyQt.exe
  Fault Module Version:	0.0.0.0
  Fault Module Timestamp:	581e0930
  Exception Code:	c0000005
  Exception Offset:	0000000000d31480
  OS Version:	6.1.7601.2.1.0.256.48
  Locale ID:	1033
  Additional Information 1:	aa8b
  Additional Information 2:	aa8b8565f92a21edea6add95009a38df
  Additional Information 3:	28ac
  Additional Information 4:	28ac02407e34c72c3170e7d5d407e38b

In order to get TYQT to load I have to disable the below drivers circled in Red
[image]

Thank you,
Scott Wegner

Koromix, 2017-02-23:
Well that's certainly a problem :) Could you try the latest development build, the application is now named TyCommander? There have been several changes in the Win32 device code since the latest release 0.8.0.

Direct links to this version:
- https://bintray.com/koromix/tytools/download_file?file_path=TyTools-0.8.0-102-g7b89d86-win32.msi
- https://bintray.com/koromix/tytools/download_file?file_path=TyTools-0.8.0-102-g7b89d86-win32.zip

xorionfe, 2017-02-23:
Thank you Neils.  I will give it a try and report back.

Best regards,
Scott Wegner

xorionfe, 2017-02-23:
>    path=TyTools-0.8.0-102-g7b89d86-win32.msi
>    path=TyTools-0.8.0-102-g7b89d86-win32.zip

Unfortunately Tycommander exhibits the same crash

Problem signature:
  Problem Event Name: APPCRASH
  Application Name: TyCommander.exe
  Application Version: 0.0.0.0
  Application Timestamp: 58a9b0ab
  Fault Module Name: TyCommander.exe
  Fault Module Version: 0.0.0.0
  Fault Module Timestamp: 58a9b0ab
  Exception Code: c0000005
  Exception Offset: 000a4930
  OS Version: 6.1.7601.2.1.0.256.48
  Locale ID: 1033
  Additional Information 1: 0a9e
  Additional Information 2: 0a9e372d3b4ad19135b953a78882e789
  Additional Information 3: 0a9e
  Additional Information 4: 0a9e372d3b4ad19135b953a78882e789

xorionfe, 2017-02-23:
if I disable the below devices then Tycommander starts fine

[image: Inline image 1]

Koromix, 2017-02-23:
I really need to produce a log file for this kind of crash. In the meantime, could you put the batch file in the ZIP below in the TyCommander directory and run it?

It will try to start TyCommander in debug mode and output the debug log in a text file "TyCommander_<DATE>_Log.txt" in the desktop directory. Then you can send me this file.

[TyCommander_Debug.zip](https://github.com/Koromix/tytools/files/796761/TyCommander_Debug.zip)

xorionfe, 2017-02-23:
> a text file "TyCommander__Log.txt" in the desktop directory. Then you can

Tried to run it but it fails to run

[image: Inline image 1]

jbliesener, 2017-02-23:
Scott, I'm following this as well. Can you try to re-upload the picture you were attaching? Didn't come through...

xorionfe, 2017-02-23:
[image]

Koromix, 2017-02-23:
Not sure why the batch code fails but this is why I hate batch... Although it could be because you run an Administrator command prompt, so it fails to find the desktop directory.

Can't test if that's the cause, no admin access on my work computer.

To output the log data, do "set TY_DEBUG=1" in the console and then execute "TyCommanderC.exe run". It'll give you the debug output in the console.

xorionfe, 2017-02-23:
Here is result

[image]

Koromix, 2017-02-23:
Thank you, that gives me a pretty good idea of what goes wrong and how to reproduce without access to these USB controllers.

I'll push a (potential) fix tonight.

xorionfe, 2017-02-23:
Thank you Niels.   If you need anything else I am around most of the day at
my home office so can test stuff.

Koromix, 2017-02-23:
I've pushed a few fixes that may or may not help with this problem... More importantly, the latest build adds a crash memory dumper to TyCommander to help me track this issue.

https://bintray.com/koromix/tytools/tytools/view#files
Get build TyTools-0.8.0-107-ge5c1e9a

Get this build, start TyCommander.exe and send me the memory dump that gets saved in "C:\Users\<NAME>\AppData\Local\CrashDumps". Well, if it works. It does on my computer but so did the batch file...

xorionfe, 2017-02-24:
Crashed again

Here are the dmp files.   Looks like it created two of them.

Scott

Koromix, 2017-02-24:
Sorry, the two files are not in your post. GitHub probably failed with the mail attachments.

You can send me the files directly if you prefer: niels.martignene@gmail.com

Koromix, 2017-02-24:
The crash should be fixed in TyTools-0.8.0-110-g9fdf575.

Can you confirm that first, the crash does not happen anymlore and two, that TyCommander detects Teensy devices when they are plugged in the USB ports controlled by the VIA and ASMedia host controllers?

xorionfe, 2017-02-24:
Thank you Niels.  I will try it again after work tonight when I get home
and report back.

xorionfe, 2017-02-24:
Just got home.  It is working!   and yes, I am able to use no matter which
of the ports I plug into.  It seems to be working perfect.   Thank you.
Without this program managing all the Teensys I am about to have running
this sim would be difficult.  I appreciate you making it and fixing it to
work with these drivers.   If there is somewhere I can send a little
donation please let me know.

Scott

Koromix, 2017-02-25:
Glad to see it works! Thanks for you help in fixing this issue. I will ship the 0.8.1 release soon, mainly to include this fix.

I have a Paypal page up if you want: https://www.paypal.me/nielsmartignene
But I don't even know how / if this page works. The Paypal account does work, it is linked to niels.martignene@gmail.com.

xorionfe, 2017-02-25:
As an FYI, the upload functionality is not working.  It hangs.  I then
reset the Teensy in question by pressing its reset button and it appears to
load then errors out.   Will wait for your 8.1 release and try again.

Scott

Koromix, 2017-02-25:
Does the serial monitor work correctly, at least? And does upload work on a USB 2.0 port, if you have one?

USB 3.0 controllers are annoying on Windows < 10, because unlike USB 2.0 controllers they each use a specific constructor driver... and those tend to exhibit various bugs. Simply looking at them the wrong way can cause weird stuff.

I've used your donation to buy a VIA PCI USB 3.0 controller to test stuff on it. It'll be here in a few days, so I can test TyCommander on it.

xorionfe, 2017-02-25:
I just pulled the via hub out and moved my connections over onto the main board and seems to be fine now uploading

Koromix, 2017-02-25:
I'm not surprised. I've known for a long time that USB 3.0 controllers on Windows < 10 are hit and miss with this program.

I've found both a VIA and an ASMedia PCI-E to USB 3.0 cards, both pretty cheap, to test and fix the issue. They should arrive next week. Thanks for the donation!

xorionfe, 2017-02-25:
It wasn't only with your program.  I was seeing some other usb problems with the teensys in x-plane.   They were also connecting through usb 2.0 hubs.   Moving the hubs and removing the via card seems to have resolved for now.    I guess it is all about the correct combination and maybe some hubs are a little sensitive to the type of usb chipset they are connecting to.

Koromix, 2017-05-28:
I've made a new release, TyTools 0.8.5. There's not much new beyond UI changes since we've fixed this issue.

The (lightweight) testing I've done with the boards I've received has not revealed any issue so I'm closing this for now.

closed 2017-05-28 by Koromix