Koromix/tytools #13: switch through serial history with up/down arrow keys

gschora, 2017-03-06:
It would be awesome if there was a possibility to switch between the last entered serial commands from the history with the arrow keys. So you wont have to grab the mouse if you are entering lots of commands.

Btw this is really awesome! I'm currently using 8 different teensys and this tools make it sooo much easier!
Thanks!

Koromix, 2017-03-10:
This is an obvious must-have, since without it the serial monitor is painful to use.

I want to fix this pretty quickly, maybe this weekend. But I'm kind of engaged on multiple things at the time right now and time is an issue, so no guarantee.

Koromix, 2017-05-06:
Better later than never, this is supported in the latest development build "TyTools-0.8.0-141-ge0cd89f".
You can find these builds here: https://bintray.com/koromix/tytools/tytools/view#files

For UI and code simplicity I've removed the history menu, which is made mostly useless by this change.

gschora, 2017-05-07:
Great!
Will try this right away...
Thanks!

Koromix, 2017-05-08:
The most recent build also supports navigating the serial history with the mouse wheel, if you need it.

This feature seems completed to me, so I'm closing the issue. If there's a problem feel free to post a new comment or open a new issue.

closed 2017-05-08 by Koromix