Koromix/tytools #14: Would love ability to save Serial data to a log file

systronix, 2017-04-30:
It would be great to be able to save Serial data to a log file for easier post-mortem analysis. This could mean saving serial data to a log file as it comes in (that would be the nicest, and would give all the data, even if the buffer has filled and old data is lost in the app), or by selecting Serial>Options>Save to File.

If you get this ability I will test it immediately!

Thanks for a great program!
Bruce Boyes

Defragster, 2017-04-30:
This already exists! Per Teensy Ser# and looks like it may be 3 level deep in history if device comes and goes.

At least on Windows - look in %appdata% from something like::
"C:\Users\YourName\AppData\Local\Temp\TyCommander-2272860-Teensy-1.txt"

systronix, 2017-04-30:
Yeah, OK I feel dumb! I just found the reference to the log file in Serial>Options. However when I go to look at the file it has old data from a couple of days ago, and does not have the data I can see in the serial window. Then I came here to edit my post above and saw your reply. So when does data actually get written to the log file?

Defragster, 2017-05-01:
I plugged in a Teensy and went and looked and that data was there that minute while the Teensy was still connected.  Did a 'Reset' and the new text was in the same file as I looked.  Koromix can confirm when it gets fully written.

Koromix, 2017-05-01:
When a board is plugged in TyCommander opens one of "C:\Users\<YourName>\AppData\Local\Temp\TyCommander-<serial>-Teensy-<1 to 3>.txt" log file. The first non-existing file or the one with the oldest modification time gets picked. That means that the 1-3 suffix does not indicate time order, you need to look at the modification date / time.

There is a maximum size for each log file (defaults to 20 MB) and I chose to loop back to the beginning when this size is reached. The separator line contains "@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@". I don't think it is possible to truncate the beginning of a file quickly and reliably on all platforms and most softwares would not like that anyway. The alternative is to clear the whole file and start again when the maximum size is reached but I don't like that.

You can change the maximum default size in "Tools > Preferences" or the size for a specific board in the "Options" tab.

Koromix, 2017-05-01:
If this behavior does not fit your use case very well don't hesitate to make suggestions.

Here are the reasons I made it work that way (for now):
- I don't want the user to have to manually set a path for each board, hence the automatic path generation...
- ... But I don't want to waste too much space the user may not even be aware is being used. Hence the maximum size per board.
- I didn't want to deal with renaming files to reorder the 1-to-3 because of Windows, which prevents renaming if a file is in use which could be a problem with some text editors. So instead I pick the oldest file.

systronix, 2017-05-05:
Well, I am looking for the log to one of five Teensy I have currently running test code. I can see the file name in Serial Options. I go to that folder C:\Users\BAB\AppData\Local\Temp but there are 6,397 items there currently.  It might be nice to set my own log folder.

I had checked "clear on reset", but I guess that doesn't mean start a new file? What is the function of that box?

The naming convention is fine.

I like it treating the file like a ring buffer, so I always have the log of the most recent XXX bytes.

I am so grateful you wrote this. I am having more and more systems running test code 24/7 and it is really nice to have a good tool to help manage that. Being able to name each system is very useful too.

Defragster, 2017-05-06:
Clear on reset clears the GUI window text - like the IDE does - after an Upload. Might be nice to have the Log File close and cycle to another file [1,2,3] as it does that as well?

Finding the %appdata% file path is a pain - once I get there I sort by date and the newest pops to the top. But being able to have it stored to a specified spot - even in the TyComm program directory "\log" might be nice.

Koromix, 2017-05-06:
I've added some code to help here:
- The **'Clear on reset' rotates the current log file** when you reset the board. Again, since TyCommander uses the modification date, it may not always cycle in an obvious "1-2-3-4" pattern.
- There is now a **global option to change the log directory** (not per-board), you can find it in "Tools > Preferences".
- You can *click on the filename in the "Options" tab* to open it directly. You can also right click on it to copy the filename to clipboard.
- Cycle log file *from 1 to 4* instead of 1 to 3.

These changes are available in the development build **TyTools-0.8.0-140-gc66216a**, which you can find here: https://bintray.com/koromix/tytools/tytools/view#files

Koromix, 2017-05-08:
I think this issue is solved. If the changes I've made happen to be broken, you can still comment here, or you can create a new issue.

closed 2017-05-08 by Koromix