Koromix/tytools #16: Generic has No Serial Log

Defragster, 2017-05-07:
Current version 8-146 has NO serial log presented:
[image]

Koromix, 2017-05-08:
This is already covered by #10, specifically point 3. I'll post details about it there.

closed 2017-05-08 by Koromix

Defragster, 2017-05-08:
Okay - it was working before - it was a regression, glad it is back

Koromix, 2017-05-08:
Sorry for closing this issue so quickly. Actually it was not supposed to work before: the serial log is disabled when the UNIQUE flag is not set. This means the log worked by accident.

Defragster, 2017-05-08:
No problem - glad it worked before or I would have complained more :)

Thanks for the fixes and changes!  I just pulled down 0.8.0-160 and I can name my devices now!