Koromix/tytools #20: Problems opening the com port

luni64, 2017-08-23:
I'm using TyCommander from a makefile under Win10 which worked perfectly so far. However, since I installed the new version (0.8.8) I get errors  complaining that it can't open the COM port (don't have my system here so I can't reproduce the exact error message). Despite the error message it starts the bootloader (but doesn't start downloading). The next upload try works then.
I suspect some timing issue. Looks like the Bootloader takes longer for start up (or for show up) than you wait...

I estimate that the error appears at about 80% of the downloads.

Koromix, 2017-08-24:
Hi,

Could you tell me what :
- Version you were using before 0.8.8? (I published several version without announcing them in the forum so you could easily have missed them)
- Teensy model you experience this problem with?
- USB type (in the Arduino Tools or Options menu) is configured?

Can you also try a development build before I made some changes to the upload code (to make it faster), namely this one: https://bintray.com/koromix/tytools/download_file?file_path=TyTools-0.8.6-7-gfc83cc7-win32.zip

Thanks!

luni64, 2017-08-24:
Hi, 
Thanks for the quick response. Don't know exactly what version I used before, last working Installation was around June. 

I used a T3.5 USB Serial.

I'm currently traveling, can try the linked build on Monday or Tuesday.

luni64, 2017-08-30:
Hi, 
the error somehow vanished. Probably something was stuck in my system. Works now with Arduino and VS2017 (makefile) without any issues. 

Thanks for the great tool

Koromix, 2017-09-17:
Thanks for reporting back. This kind of stuff happens from time to time, whether it comes from the OS, the hardware, the board... USB is a complex beast. I'm closing this bug, feel free to open a new one if stuff goes wrong again.

(Sorry for the late answer)

closed 2017-09-17 by Koromix