Koromix/tytools #21: IHEX parse error

luni64, 2017-10-03:
_Windows 10, TyTCommander Version 0.8.8_

I get the following parse error for the attached hex file.  

[image]

Uploading the corresponding .elf file (attached) works however.  Teensy.exe uploads the hex without problems.  

[usbhost.zip](https://github.com/Koromix/tytools/files/1353376/usbhost.zip)

luni64, 2017-10-03:
I just had a look at the hex file. Line 2755 is at the end of the file. Looks like you are stumbling over the 05 Record (Start Linear Address). I have no idea why objcopy adds this record in this case and not in all others.  

Here a corresponding SO question:  [https://stackoverflow.com/questions/26295776/what-are-the-intel-hex-records-type-03-or-05-doing-in-ihex-program-for-arm
](https://stackoverflow.com/questions/26295776/what-are-the-intel-hex-records-type-03-or-05-doing-in-ihex-program-for-arm
)

closed 2017-10-06 by Koromix

Koromix, 2017-10-06:
Thanks for reporting this bug,

I've pushed a commit to fix this issue. Development build *TyTools-0.8.8-7-g0c123c2* on [BinTray](https://bintray.com/koromix/tytools/tytools/view#files) should work correctly.

luni64, 2017-10-06:
Works, 
Thanks for fixing

commit 0c123c2a (2017-10-06): Fix handling of records 03 and 05 in Intel HEX files

Fixes #21.