Koromix/tytools #22: Binary output in display - causes clipboard issues with cut and paste

KurtE, 2017-11-08:
I have had a few instances recently where, some text was output that was displayed in the text output output window.  

I then try to do a select all command and then a copy.  When I go to some other window (like sublime text) window, and try to do a paste.  Either the clipboard was not updated at all with the COPY command in Tycommander or a subset of the text is on the clipboard. 

I am not sure yet what characters are causing the issue, but it has happened several times. 

I am running 0.8.8-7-g0c123c2 on a Windows 10 64 bit machine. 

Kurt
P.S. - Great tool!

KurtE, 2017-11-09:
Thought I would mention some characters that I saw this with.  I am testing USB Host code on T3.6 and trying to do communications with AX servos (or controllers).  The test app echoes everything it receives on the USB Host Serial port to the Serial object... The one message I see that output sort of garbage to output window which if I include it in a copy of the clipboard causes the problem: 

    rx: FF FF FD 02 08 F8 FF FF FD 04 00 82 08 74 

Note: this hex bytes are output as their actual binary values. (not the rx:)

Side note: I was trying to replicate an issue where I hung TyCommander and everything I typed showed some messages in Red at bottom of window (not in scroll area).

Figured it out.  In my test program I have it set, such that if I type:

    #1000000 

Or any number it closes the USB host serial port and reopens it at that baud number.  But one time I mistyped and did: @1000000 

It complained that the file 1000000 does not exist and then the window was functional.

Koromix, 2017-11-20:
Hi,

The copy-paste bug may be caused by NUL bytes in the text, I'll have check this out this week or next week-end.
Could be a Qt bug, hopefully not but I do static Qt builds so at least I can fix it... if it does not go too deep in Qt, because the code behind the text widgets is hard to follow / kind of a mess.

The '@' is not so much a bug as a hidden (possibly misguided) feature, '@filename' is treated as a command to send the data in 'filename' to the board. Hence the error message, because the '1000000' file does not exist.

I've changed it in the latest developement build (TyTools-0.8.8-11-g0f26704), now there are multiple '@' commands:
- '@send_file filename' = send file data
- '@send text' = send text (somewhat redundant, but you can use this to send text that starts with '@', otherwise it'll be treated as a command)

KurtE, 2017-11-20:
Thanks,

 

As for the @100000 trying to open file name, which does not exist, give an error.  As you mentioned sounds like a feature.  The only issue that I saw was I did not see any way out of this error condition, other than exit the app. 

 

Will try out your new build and continue to have fun.  

 

Thanks for your work here!!!

closed 2018-09-22 by Koromix