Koromix/tytools #24: On macOS High Sierra / APFS formatted disk, platform.txt is written with wrong path

Theremingenieur, 2018-02-27:
During the integration procedure, one has to select the path towards the Arduino/Teensyduino installation. Usually, one picks /Applications/Arduino.app. But then, in platforms.txt, it writes a strange path like 
_tools.teensyloader.cmd.path=/private/var/folders/cq/bj9dgxhj3c50b4xprr6d5qqh0000gn/T/AppTranslocation/35AB90F4-8443-4939-9668-34119873A20A/d/TyCommander.app/Contents/MacOS/TyCommander_
which is wrong and will prevent TyCommander from launching after compiling.
Each time, I'm forced to edit platforms.txt manually to set the path to _/Applications/TyCommander.app/Contents/MacOS/TyCommander_ which will make it work.

Theremingenieur, 2018-04-24:
Might have to do with the "app relocation" security thing in newer macOS versions

Koromix, 2018-04-24:
Hi,

Yes, seems to be the case.
I've been looking for a few 'cheap' enough (and not too annoying to get) code signing certificate but there's not much available to justify the cost and annoyance.
Maybe I will add a warning in TyTools instead, in the mean time.

Just in case, do you run TyTools from /Applications or somewhere else (such as a mounted volume)?

Niels

Theremingenieur, 2018-04-24:
From /Applications. I've a well organized OS 😃

Theremingenieur, 2018-05-18:
Looks like it has been solved? Installed the newest TD 1.42 beta 5 and at the same occasion the last TyCommander build from bintray and the problem seems to be gone. Platform.txt was rewritten with the correct application path.

Theremingenieur, 2018-06-11:
Seems to be solved. Closing the issue for now.

closed 2018-06-11 by Theremingenieur