Koromix/tytools #25: tcsetattr() failed on '/dev/cu.usbmodem#####': Invalid argument

dkadish, 2018-03-01:
I have been getting an error for the last week or two using TyQt on OSX High Sierra (10.13.3) with a Teensy 3.6. I've tried TyQt 0.8.0 and TyCommander 0.8.8 with the same result.

The serial monitor works on the initial plug-in, but then after trying to upload new firmware I get the tcsetattr failure message. This message also occurs on subsequent attempts to start the serial monitor.

Any ideas? I'm happy to play around with it a little but I'm not sure where to start debugging.

Full log:
```
Add serial device '6fae6ce4ae35' on iface 1
  - USB VID/PID = 16c0:0483, USB location = usb-20-1
  - USB manufacturer = Teensyduino, product = USB Serial, S/N = 3377180
tcsetattr() failed on '/dev/cu.usbmodem3377181': Invalid argument
Remove device '6fae6ce4ae35'
Add HID device '6fe459b990e2' on iface 0
  - USB VID/PID = 16c0:0478, USB location = usb-20-1
  - USB manufacturer = (none), product = (none), S/N = 00052736
  - HID usage page = 0xff9c, HID usage = 0x22
Identified 'Teensy 3.6' with usage value 0x22
Remove device '6fe459b990e2'
Add serial device '6fe69088e855' on iface 1
  - USB VID/PID = 16c0:0483, USB location = usb-20-1
  - USB manufacturer = Teensyduino, product = USB Serial, S/N = 3377180
tcsetattr() failed on '/dev/cu.usbmodem3377181': Invalid argument
tcsetattr() failed on '/dev/cu.usbmodem3377181': Invalid argument
Remove device '6fe69088e855'
Add serial device '6ff0b73242c0' on iface 1
  - USB VID/PID = 16c0:0483, USB location = usb-20-2
  - USB manufacturer = Teensyduino, product = USB Serial, S/N = 3377180
Remove device '6ff0b73242c0'
Add serial device '6ff40b4ef80a' on iface 1
  - USB VID/PID = 16c0:0483, USB location = usb-20-2
  - USB manufacturer = Teensyduino, product = USB Serial, S/N = 3377180
[upload@3377180-Teensy] Uploading to board '3377180-Teensy' (Teensy 3.6)
[upload@3377180-Teensy] Triggering board reboot
Remove device '6ff40b4ef80a'
Add HID device '700105ad752a' on iface 0
  - USB VID/PID = 16c0:0478, USB location = usb-20-2
  - USB manufacturer = (none), product = (none), S/N = 00052736
  - HID usage page = 0xff9c, HID usage = 0x22
Identified 'Teensy 3.6' with usage value 0x22
[upload@3377180-Teensy] Firmware: firmware.elf
[upload@3377180-Teensy] Flash usage: 63 kiB (6.1%)
[upload@3377180-Teensy] Sending reset command
Remove device '700105ad752a'
Add serial device '70014cac22a1' on iface 1
  - USB VID/PID = 16c0:0483, USB location = usb-20-2
  - USB manufacturer = Teensyduino, product = USB Serial, S/N = 3377180
tcsetattr() failed on '/dev/cu.usbmodem3377181': Invalid argument
```

Theremingenieur, 2018-03-28:
Strange... can’t reproduce that with T3.6 and High Sierra 10.13.4 beta 7.
Does that only happen with a specific firmware? Did you try a known working one like the famous blink sketch?
How did you compile your firmware? Using the Teensyduino toolchain or a different approach?

Koromix, 2018-04-24:
First test, is the Teensy plugged directly or through an USB hub?
If it's the latter option, can you try without the hub?

getSurreal, 2018-07-22:
I'm having the same issue with Teensy 3.1 and RPi and no USB hub.  
`tcsetattr() failed: Invalid argument`
I just started this setup yesterday and it was working at least a couple of dozen times without the issue.  Today I put the upload command followed by the monitor command into a script and that's when it started.  I thought maybe it was trying to monitor too soon after the upload so I tried some various sleep times in between.  Even with a 5 second sleep it's still happening.  It would consistently get that error every second time I executed it and I would have to reboot after getting the error.  (Unplugging and plugging may have worked, but it was not next to me so a remote reboot was easier)

I removed the monitor from the script and manually ran it and I stopped getting the error, but I definitely don't wait more than 5 seconds.  It doesn't make sense to me, but I thought I would share my results.  I just did it about 6 times in a row again without the error.

Koromix, 2018-09-23:
Hi,
I'll try to reproduce this issue today on my Raspberry Pi.

Koromix, 2021-01-19:
Several bugs have been fixed since this report was made. I can't reproduce this, so closing the issue.

closed 2021-01-19 by Koromix

koraysels, 2022-01-04:
I'm having this issue since today.. don;t know what changed.. tried the new tyupdater version but that does also not work. 
`tcsetattr() failed on '/dev/cu.usbmodem#####': Invalid argument`

Htis is wuith a usb cable that worked before perfectly.. 

[image]

Koromix, 2022-01-07:
What version of macOS is it? And what is the version of TyTools?

koraysels, 2022-01-07:
latest tytools, big sur . 

but i tried the same on my hackintosh and that did work, so there porbably is a driver issue of some sorts on my macbook…