Koromix/tytools #26: T3.5 now has 256K of memory - Error on uploads

KurtE, 2018-05-12:
With the new beta (https://forum.pjrc.com/threads/52090-Teensyduino-1-42-Beta-4), 
The Teensy 3.5 now has 256k of memory instead of 192 ;D 

But now TyCommander shows an error if you try to upload a new program to the T3.5

Koromix, 2018-05-13:
Thanks for the report,

It should be fixed in the latest development build (TyTools-0.8.8-24-g569e504).
You can find it on BinTray: https://bintray.com/koromix/tytools/tytools#files

Feel free to reopen the bug if it does not work for some reason.

Koromix, 2018-05-13:
Well actually, since I don't have a Teensy 3.5, can you confirm if it works?
If it does I'll make a new official release.
Thanks!

KurtE, 2018-05-13:
Thanks!

 

It appears to work on my T3.5 I was testing with!

Defragster, 2018-05-13:
I just downloaded and integrated the above new TyComm and sent a simple sketch to my T_3.5 with no trouble.

Would have tested sooner - but 'somebody' :) told me it wouldn't matter :)

Niels - Good Work on TyCommander!  PJRC is advancing the multi use Teensy case with new Teensy_Sermon [ multiple IDE's can now open and program multiple Teensys ] - but your work focused on the Loader/SerMon combo is still much better!  It doesn't get confused with multiple instances of SerMon touching the active devices - and the GUI support with RESET and Bootloader buttons is awesome.  Also the implied SCROLL stop as you have it is awesome - versus the 'checkbox' autoscroll copied from Arduino SerMon.

Defragster, 2018-05-13:
Niles - TyComm USB SerMon is a bit faster than the new PJRC scheme in general.

I've got a T_3.6 on USB3 port pushing ~28,000 messages into TyComm and using the Teensy_sermon it only pulls fast enough to get ~27,000 - not a huge difference - but no slower.  TyComm UI does get a bit 'clunky' with 200K line buffer with 28K messages/sec - but dropping Options to 20K line buffer fixes that. On a USB2 or USB3 hub the transfer rates drop to 20K BTW - and on USB3 port a T_LC is only 17% slower at this pure print task sending ~24000 messages/sec and the new Teensy_sermon can keep up with that.

Not to mention - though here it is - the Loader interface is better too.

Also - other subtle things that make usage very handy:
>>Serial On/Off switch is awesome to release the device - IDE and PJRC don't offer that.
>> When device goes offline TyComm allows select and cut - the other two block that UI

luni64, 2018-05-14:
> Anyway, glad to see this little software is still useful :]

This 'little software' is one of the biggest performance boosters I came across the last years. I do a lot of development involving more than one teensy and can't remember how I ever could work without tytools :-) 

Thanks for that very useful piece of software.

Defragster, 2018-05-14:
I don't see how to RE-OPEN ….

The Beta 4 fix shifts RAM wrongly and causes runtime errors.  RAM alignment is off by 4 bytes from 262140 goes to 262136.

FrankB made a fix that results in:
[upload@2114150-Teensy] Firmware 'sketch_may13a.ino.TEENSY35.hex' is not compatible with '2114150-Teensy'

The changes he uploaded are to Boards.txt ::
teensy35.upload.maximum_data_size=262136

and to \hardware\teensy\avr\cores\teensy3\mk64fx512.ld ::
	RAM  (rwx) : ORIGIN = 0x1FFF0000, LENGTH = 262136

Koromix, 2018-05-14:
New build will be up soon. I'll wait for Teensyduino 1.42 final to make a new release :)

Koromix, 2018-05-14:
I want to add a new per-board option for this release, "Ignore firmware version check", to work around this issue in case (when?) this happens again.

closed 2018-05-14 by Koromix

Defragster, 2018-05-14:
downloading to test ...

commit 569e5048 (2018-05-13): Recognize firmwares built for Teensy 3.5 with RAM = 256K

Fixes #26.
commit d93a283f (2018-05-14): Fix Teensy 3.5 firmware recognition again

Fixes #26.