Koromix/tytools #32: tycmd list does not show Teensy submodel

ttlappalainen, 2018-09-05:
On your readme you define that tycmd list should show:
add 34130@usb-1-2 Teensy 3.1
add 29460@usb-4-2 Teensy

for me it shows just:
add 1241320-Teensy Teensy (USB Serial)
add 3677880-Teensy Teensy (USB Serial)

TyCommander shows right Teensy 3.6 or Teensy 3.2.

Should tycmd also show submodel 3.x?

Koromix, 2018-09-23:
Hi,

The main problem is that, until recently, the submodel could only be known while the Teensy was in bootloader mode.
TyCommander uses a cache, stored on disk, to retain the precise model for each board (using the serial number) after is has seen the bootloader info at least once.
I've never added the ability to use this cache to the command-line version.

However, for boards programmed with Teensyduino 1.42+, there is a way to get the model without the bootloader. I've published a new version (0.8.10) that can use this feature, both in TyCommander and tycmd.

Koromix, 2018-09-23:
Still, it's not perfect because the Teensy 3.2 can be detected as 3.1 (software-side they are identical).
I'm keeping this open because I'd like to add cache support to tycmd eventually.

ttlappalainen, 2018-09-24:
Thanks, I'll check that.

But does that actually make any difference, if I use Platform IO for creating hex files? 

On the other hand can I see model, If I have empty Teensy on some port?

Koromix, 2018-09-25:
I just checked and PlatformIO includes Teensyduino 1.41, so it won't work.

You can either ask whoever maintains the Teensy package for PlatformIO to update it, or overwrite the teensy3 core code with newer ones.

ttlappalainen, 2018-09-25:
OK.

But how does it actually read device info? Does it somehow read the flash content.

Koromix, 2018-09-28:
There are two ways to get version information from a Teensy:
- in bootloader mode, the usage value in the HID report descriptor has a specific value for each model
- since Teensyduino 1.42, the bcdDevice in the USB device descriptor has a specific value for each model (in most modes)

ttlappalainen, 2018-10-18:
Thanks. Platform IO has now Teensyduino 1.44 so it seem to return model information. 

Two more questions:
- If I have board running and no serial to open it, should it also show "upload" under "capabilities".
- On TyCommander there is also "bootload" command to force device to bootloader. Could that be also on TyCmd? The difference is that it could then return device interface "halfkay" ID.

ttlappalainen, 2018-10-18:
An other question:
On TyCmd JSON output you return e.g.  "interfaces": [["Serial", "\\.\COM14"]] . Shouldn't that actually be:
 "interfaces": [{"Serial":"\\.\COM14"}] ?

ttlappalainen, 2018-10-18:
Also \ should be escaped -> "interfaces": [{"Serial":"\\\\.\\COM14"}]

Koromix, 2019-09-14:
Better late than never I hope.

There may be (at least theoretically) duplicate interface names, hence the use of nested arrays instead of JSON objects for interfaces.
OTOH, I've fixed JSON string escaping in the latest commit. You can find a Win64 build here:
https://koromix.dev/files/tytools

You can use "tycmd reset -b" to go in bootloader mode.

closed 2019-09-14 by Koromix