Koromix/tytools #33: Tycmd compatibility with Unity3D

HiruniNuwanthika, 2018-09-12:
I was able to get some position data printed on tycmd. Is there a way to integrate that data into unity3D?
I have tried with a c# code. But the problem occurs when both Tycmd and Unity accessing the same port.
Please help me in solving this problem. Thank you!!!

Theremingenieur, 2018-09-12:
Serial ports are always exclusive for one to one connections. So, on the PC side, it’s either Tycmd, or a “classic” terminal program, or another program accessing the serial port.
The most common way if you have an embedded MCU sending serial data to another application is to use two serial ports. One which sends data to your app and another which sends debug data and information to a serial monitor like Tycmd.

HiruniNuwanthika, 2018-09-12:
Thank you for your answer..
I have tried only with a c# code using serial port. But I received a timed out error. 
Is there a way to access data from Unity instead of printing on tycmd?

Theremingenieur, 2018-09-12:
If your MCU (you even don’t write which one you use, Teensy 3.x?) sends data over serial, it can basically be read and be captured by whatever program which can access and read the serial port. If you see data in Tycmd, that’s the proof that your MCU sends data correctly. Next step would be a check with a known working terminal program which naturally could only access the serial port if Tycmd is closed and the serial port is free. In a third step, you might try to write your own program which could read serial data, still under the condition that nothing else blocks the port.
Since there have been people acquiring serial data from a Teensy with Python and MATLAB, it should also be possible with other stuff. But this is definitely not a Tycmd question or issue! You are asking in the wrong place.
You register on the Teensy forums https://forum.pjrc.com and use the search function there to find working examples. Or you ask in a C# developer forum to learn how to read data from a serial port. I can’t help you for that, as far as I know, C# is for Windows, which is not my world.

HiruniNuwanthika, 2018-09-12:
Thank you very much for your support...

Koromix, 2018-09-22:
Seems like this issue is resolved (thanks Theremingenieur), I'm closing it.

closed 2018-09-22 by Koromix