Koromix/tytools #36: Binaries for T4 branch

luni64, 2019-01-12:
Hi, are there already binaries for the T4 branch?

Koromix, 2019-01-23:
For Win32: https://ci.appveyor.com/api/buildjobs/3l4ts50tikgn0gbr/artifacts/TyTools-0.8.10-2-g1ed1eff-win32.zip
For Win64: https://ci.appveyor.com/api/buildjobs/mucrlny6ruoiaqp7/artifacts/TyTools-0.8.10-2-g1ed1eff-win64.zip

It should refuse to talk to the bootloader (so no upload and no complete reset) unless you pass in an environment variable: "TYTOOLS_EXPERIMENTAL_BOARDS=1" (or any value, it just needs to exist).

I don't recomment trying to upload. I haven't tested it, and I don't know if the Teensy 4 is brickable or not. Reset is probably fine, even though you need to set the environment variable to do it.

luni64, 2019-01-23:
Thanks a lot. 
Since someone needed to test it at the end I did :-)  -> Works perfectly and increases the fun factor significantly.

Koromix, 2019-01-23:
Great, thanks for the sacrifice! I would have tested it eventually, but since I don't have time or will to play with a Teensy 4 I did not ask for a beta board.

The usage info should be fixed in the most recent commit. Builds are here:
- Win32: https://ci.appveyor.com/api/buildjobs/mluoybcxh1ju0iyx/artifacts/TyTools-0.8.10-3-g0f9a80e-win32.zip
- Win64: https://ci.appveyor.com/api/buildjobs/ibte8o79ucy355uj/artifacts/TyTools-0.8.10-3-g0f9a80e-win64.zip

luni64, 2019-01-23:
Looks better:

```
> Executing task: C:/toolchain/make/make.exe uploadTy -j -Otarget <

       upload@1000040-Teensy  Uploading to board '1000040-Teensy' (Teensy 4.0)
       upload@1000040-Teensy  Triggering board reboot
       upload@1000040-Teensy  Firmware: QUAD.hex
       upload@1000040-Teensy  Flash usage: 13 kiB (0.8%)
       upload@1000040-Teensy  Uploading...
       upload@1000040-Teensy  Sending reset command
```
Not that it would be of any importance to me, but  512kB * 0.8% = 4kB. Which doesn't fit to the reported 13kB. Or am I misinterpreting the numbers somehow?

FrankBoesing, 2019-01-23:
Don't we have ~1.5MB flash?

luni64, 2019-01-23:
Ah, mixed it up with the RAM size. All good then.

Defragster, 2019-01-23:
Good work Koromix - it works on my T4. I noted on the version you gave me the other week that Reset and Bootloader buttons worked fine - and they still do - and now I've tested upload once so far after doing Integrate it looks good.

BTW - have you any ideas how the TyComm code might handle faster SerMon output from T4?  It is going from a 12 Mbit connect to a device with 480 Mbit connect speed.

Koromix, 2019-01-24:
No, it won't handle it. Some changes may be needed to the read code in libty, but that should not be a big problem.

The main problem comes from the Qt text widget, which is too slow (it's made for editing text, not fast append-only viewing). For faster speeds, we need a custom widget. It's not a huge task, because we don't need that much (and it would be a good opportunity to support a subset of VT-100 escape codes), but I'm not sure I will have time for this anytime soon.
Maybe one lonely week-end I will work on that, but no promise :)

luni64, 2019-01-24:
`and it would be a good opportunity to support a subset of VT-100 escape codes`

That would be fantastic. It would allow to use tyCommander as a nice UI for a lot of small applications.

Defragster, 2019-01-24:
Nice there may be a way to speed improvements.  Right now it seems a few byte/sec slower than PJRC's alternate Teensy Sermon in the IDE. The T4 is passing what looks like 1.5 Mbits/sec - up from the old max 1.2 if I read my test right and it was lossless.

NOTE: I did Integrate of the T4 capable TyComm - got my T4 in a bad way again - not sure what the cause was.  It happened similarly before - and I can't say it was not the day I tried the early T4 TyComm doing only reset/bootloader on the T4 … if so it seems it is Teensy_Ports getting confused or maybe the T4 does some extra stuff feeding the red led tied to the bootloader chip? Very unclear but it in both cases I was using TyComm to upload to T_3.1's the Echo of Serial debug with the T4 online. In both cases it seemed to only clear with the T4 15 sec reset. It was not cleared up with Windows restarts. Before the reset the T4 would not come online if there was another Teensy 3.1 plugged in? And the T4 Button was not doing anything?  T_Loader was closed when using TyComm for upload.

Sorry - nebulous notes - I don't know how T4 went odd - but IIRC it was weeks of hundreds of good uploads and then messed up when I used TyComm to reprogram T_3.1 as serial>USB echo converters, changing their baud or adding LED blink. Not sure if anyone else is doing this? Need to update is rare - unless using another Teensy to test code against the T4 and trying to use TyComm instead of a second IDE instance - or IDE and changing the board.

closed 2019-03-25 by Koromix