Koromix/tytools #42: T4 doesn't work

luni64, 2019-08-31:
Just tried using TyCommander with T4  which didn't work. Any plans to update?

Defragster, 2019-08-31:
The current TyCommander is : Support Teensy 4 beta boards (experimental)

When you try to talk to the T4 it shows an error note saying an environment VAR must be set - at least on WINDOWS it is :: TYTOOLS_EXPERIMENTAL_BOARDS=1

Hopefully in the next week koromix intends to get back to TyCommander for T4 now that he has a board and it is released.

luni64, 2019-08-31:
@Defragster I know that TYTOOLS_EXPERIMENTAL_BOARDS=1 enables support for the T4Beta. Are you saying that it should also work for the released T4? At least here it doesn't. 

```
Waiting for user selection
       upload@6239370-Teensy  Uploading to board '6239370-Teensy' (Teensy)
       upload@6239370-Teensy  Triggering board reboot
       upload@6239370-Teensy  Board '6239370-Teensy' has disappeared
make.exe: *** [makefile:166: uploadTy] Error 1
The terminal process terminated with exit code: 2
```

Defragster, 2019-08-31:
Yes, it works on the released boards. For programming they were unchanged from the last Beta boards.

>only thing that comes to mind: Make sure that TeensyLoader is not active.

I had all the Beta units and even got pre-release first production board and used TyCommander on them to good effect … when that ENV VAR was set.

Koromix, 2019-09-11:
Hi,

Thanks for the report.
Could you try to upload code to your T4 board with TyCommander, and then send me the debug log ("View > Show Application Log > Full log)?

luni64, 2019-09-11:
Here the Log. 
```
Listing USB host controllers and root hubs
Found root USB hub 'USB\ROOT_HUB20\4&17921545&0' with ID 0
Found root USB hub 'USB\ROOT_HUB\4&29C55406&0' with ID 1
Found root USB hub 'USB\ROOT_HUB\4&327FB83B&0' with ID 2
Found root USB hub 'USB\ROOT_HUB\4&245D2815&0' with ID 3
Found root USB hub 'USB\ROOT_HUB\4&C3EEE0C&0' with ID 4
Found root USB hub 'USB\ROOT_HUB\4&1A617E32&0' with ID 5
Found root USB hub 'USB\ROOT_HUB\4&25AE9D96&0' with ID 6
Found root USB hub 'USB\ROOT_HUB20\4&8FC751&0' with ID 7
Examining device node 'ACPI\PNP0501\1'
Unknown device type for 'ACPI\PNP0501\1'
Examining device node 'USB\VID_16C0&PID_0483\6239370'
Found port number of 'USB\VID_16C0&PID_0483\6239370': 2
Found port number of 'USB\VID_0409&PID_0059\6&21374381&0&4': 4
Found port number of 'USB\VID_0409&PID_0059\5&2B739175&0&1': 1
Found controller ID for 'USB\ROOT_HUB20\4&8FC751&0': 8
Add serial device 'USB\VID_16C0&PID_0483\6239370' on iface 0
  - USB VID/PID = 16c0:0483, USB location = usb-8-1-4-2
  - USB manufacturer = Teensyduino, product = USB Serial, S/N = 6239370
Examining device node 'USB\VID_046D&PID_C03D\6&EEEB412&0&2'
Device 'USB\VID_046D&PID_C03D\6&EEEB412&0&2' has no 'PortName' registry property
Ignoring duplicate HID collection device 'HID\VID_045E&PID_07F8&MI_01&COL02\8&6D675B7&0&0001'
Examining device node 'HID\VID_045E&PID_07F8&MI_01&COL01\8&6D675B7&0&0000'
Found port number of 'USB\VID_045E&PID_07F8\6&EEEB412&0&3': 3
Found port number of 'USB\VID_05E3&PID_0608\5&2231C974&0&3': 3
Found controller ID for 'USB\ROOT_HUB20\4&17921545&0': 1
Ignoring duplicate HID collection device 'HID\VID_046D&PID_0A29&MI_03&COL02\7&718B205&0&0001'
Examining device node 'HID\VID_046D&PID_0A29&MI_03&COL01\7&718B205&0&0000'
Found port number of 'USB\VID_046D&PID_0A29\000D44E29A08': 2
Found controller ID for 'USB\ROOT_HUB\4&245D2815&0': 4
Examining device node 'USB\VID_045E&PID_07F8&MI_00\7&B5619C&0&0000'
Device 'USB\VID_045E&PID_07F8&MI_00\7&B5619C&0&0000' has no 'PortName' registry property
Examining device node 'USB\VID_045E&PID_07F8&MI_01\7&B5619C&0&0001'
Device 'USB\VID_045E&PID_07F8&MI_01\7&B5619C&0&0001' has no 'PortName' registry property
Examining device node 'USB\VID_046D&PID_0A29&MI_03\6&2CAB664B&0&0003'
Device 'USB\VID_046D&PID_0A29&MI_03\6&2CAB664B&0&0003' has no 'PortName' registry property
Unknown Teensy model with bcdDevice value 0x279
[upload@6239370-Teensy] Uploading to board '6239370-Teensy' (Teensy)
[upload@6239370-Teensy] Triggering board reboot
Received removal notification for device 'USB\VID_16C0&PID_0483\6239370'
Remove device 'USB\VID_16C0&PID_0483\6239370'
Received removal notification for device 'USB\VID_16C0&PID_0483\6239370'
Received arrival notification for device 'USB\VID_16C0&PID_0478\00098541'
Examining device node 'USB\VID_16C0&PID_0478\00098541'
Device 'USB\VID_16C0&PID_0478\00098541' has no 'PortName' registry property
Received arrival notification for device 'HID\VID_16C0&PID_0478\8&2F77DAF&0&0000'
Examining device node 'HID\VID_16C0&PID_0478\8&2F77DAF&0&0000'
Found port number of 'USB\VID_16C0&PID_0478\00098541': 2
Found port number of 'USB\VID_0409&PID_0059\6&21374381&0&4': 4
Found port number of 'USB\VID_0409&PID_0059\5&2B739175&0&1': 1
Found controller ID for 'USB\ROOT_HUB20\4&8FC751&0': 8
Add HID device 'HID\VID_16C0&PID_0478\8&2F77DAF&0&0000' on iface 0
  - USB VID/PID = 16c0:0478, USB location = usb-8-1-4-2
  - USB manufacturer = (none), product = (none), S/N = 00098541
  - HID usage page = 0xff9c, HID usage = 0x24
Unknown Teensy model with usage value 0x24
Unknown Teensy model with bcdDevice value 0x105
[upload@6239370-Teensy] Board '6239370-Teensy' has disappeared
```

luni64, 2019-09-11:
Sorry, the above log was for **v0.8.10**. Here the log for 0.8.11. TYTOOLS_EXPERIMENTAL_BOARDS set to 1
``` 
Listing USB host controllers and root hubs
Found root USB hub 'USB\ROOT_HUB20\4&17921545&0' with ID 0
Found root USB hub 'USB\ROOT_HUB\4&29C55406&0' with ID 1
Found root USB hub 'USB\ROOT_HUB\4&327FB83B&0' with ID 2
Found root USB hub 'USB\ROOT_HUB\4&245D2815&0' with ID 3
Found root USB hub 'USB\ROOT_HUB\4&C3EEE0C&0' with ID 4
Found root USB hub 'USB\ROOT_HUB\4&1A617E32&0' with ID 5
Found root USB hub 'USB\ROOT_HUB\4&25AE9D96&0' with ID 6
Found root USB hub 'USB\ROOT_HUB20\4&8FC751&0' with ID 7
Examining device node 'ACPI\PNP0501\1'
Unknown device type for 'ACPI\PNP0501\1'
Examining device node 'USB\VID_16C0&PID_0483\6239370'
Found port number of 'USB\VID_16C0&PID_0483\6239370': 2
Found port number of 'USB\VID_0409&PID_0059\6&21374381&0&4': 4
Found port number of 'USB\VID_0409&PID_0059\5&2B739175&0&1': 1
Found controller ID for 'USB\ROOT_HUB20\4&8FC751&0': 8
Add serial device 'USB\VID_16C0&PID_0483\6239370' on iface 0
  - USB VID/PID = 16c0:0483, USB location = usb-8-1-4-2
  - USB manufacturer = Teensyduino, product = USB Serial, S/N = 6239370
Examining device node 'USB\VID_046D&PID_C03D\6&EEEB412&0&2'
Device 'USB\VID_046D&PID_C03D\6&EEEB412&0&2' has no 'PortName' registry property
Ignoring duplicate HID collection device 'HID\VID_045E&PID_07F8&MI_01&COL02\8&6D675B7&0&0001'
Examining device node 'HID\VID_045E&PID_07F8&MI_01&COL01\8&6D675B7&0&0000'
Found port number of 'USB\VID_045E&PID_07F8\6&EEEB412&0&3': 3
Found port number of 'USB\VID_05E3&PID_0608\5&2231C974&0&3': 3
Found controller ID for 'USB\ROOT_HUB20\4&17921545&0': 1
Ignoring duplicate HID collection device 'HID\VID_046D&PID_0A29&MI_03&COL02\7&718B205&0&0001'
Examining device node 'HID\VID_046D&PID_0A29&MI_03&COL01\7&718B205&0&0000'
Found port number of 'USB\VID_046D&PID_0A29\000D44E29A08': 2
Found controller ID for 'USB\ROOT_HUB\4&245D2815&0': 4
Examining device node 'USB\VID_045E&PID_07F8&MI_00\7&B5619C&0&0000'
Device 'USB\VID_045E&PID_07F8&MI_00\7&B5619C&0&0000' has no 'PortName' registry property
Examining device node 'USB\VID_045E&PID_07F8&MI_01\7&B5619C&0&0001'
Device 'USB\VID_045E&PID_07F8&MI_01\7&B5619C&0&0001' has no 'PortName' registry property
Examining device node 'USB\VID_046D&PID_0A29&MI_03\6&2CAB664B&0&0003'
Device 'USB\VID_046D&PID_0A29&MI_03\6&2CAB664B&0&0003' has no 'PortName' registry property
Unknown Teensy model with bcdDevice value 0x279
[upload@6239370-Teensy] Uploading to board '6239370-Teensy' (Teensy)
[upload@6239370-Teensy] Triggering board reboot
Received removal notification for device 'USB\VID_16C0&PID_0483\6239370'
Remove device 'USB\VID_16C0&PID_0483\6239370'
Received removal notification for device 'USB\VID_16C0&PID_0483\6239370'
Received arrival notification for device 'USB\VID_16C0&PID_0478\00098541'
Examining device node 'USB\VID_16C0&PID_0478\00098541'
Device 'USB\VID_16C0&PID_0478\00098541' has no 'PortName' registry property
Received arrival notification for device 'HID\VID_16C0&PID_0478\8&2F77DAF&0&0000'
Examining device node 'HID\VID_16C0&PID_0478\8&2F77DAF&0&0000'
Found port number of 'USB\VID_16C0&PID_0478\00098541': 2
Found port number of 'USB\VID_0409&PID_0059\6&21374381&0&4': 4
Found port number of 'USB\VID_0409&PID_0059\5&2B739175&0&1': 1
Found controller ID for 'USB\ROOT_HUB20\4&8FC751&0': 8
Add HID device 'HID\VID_16C0&PID_0478\8&2F77DAF&0&0000' on iface 0
  - USB VID/PID = 16c0:0478, USB location = usb-8-1-4-2
  - USB manufacturer = (none), product = (none), S/N = 00098541
  - HID usage page = 0xff9c, HID usage = 0x24
Unknown Teensy model with usage value 0x24
Unknown Teensy model with bcdDevice value 0x105
[upload@6239370-Teensy] Board '6239370-Teensy' has disappeared
```

Koromix, 2019-09-11:
Can you try the build here: https://koromix.dev/files/tytools/
If it works, I'll make a new release!

luni64, 2019-09-11:
Yay, I'm back in productive mode :-) 

Works perfectly, thanks a lot

Koromix, 2019-09-11:
Thanks, closing the issue :)

closed 2019-09-11 by Koromix

Koromix, 2019-09-12:
In theory (if I haven't screwed up), you can also drop the environment variable because this build does not require it.

luni64, 2019-09-13:
I meanwhile was able to test it also, Works without the EnvVar here as well.

Koromix, 2019-09-14:
Perfect, thanks :)