Koromix/tytools #43: Baudrate for USB Serial Cables

luni64, 2019-11-04:
In cases where I can not use the native Teensy USB Serial (e.g. when I use it to communicate to a PC program) I usually connect a FTDI USB-Serial cable to Serial1 for debugging. 
Since tyCmd is able to connect to the FTDI I would like to use its serial monitor for debugging. 

However, it looks like it doesn't detect the baud rate correctly. 

[image]

After I connect the cable to putty, set the correct baudrate and quit putty,  it works in tyCmd as well:

[image]

Any chance to add  baudrate setting / autodetect to generic boards?

Koromix, 2019-11-29:
In the latest commit, I've added a per-board serial rate setting available in the "Options" tab. Like other settings, TyTools remembers it.
I haven't tested it with a non-Teensy board but it should work the same.

You can find the corresponding Win64 build here: https://koromix.dev/files/tytools/

luni64, 2019-11-29:
Thanks, this kind of works. Changing the baudrate to the correct value only works after power cycling (plug out/in) of the usb cable. Looks like it doesn't apply the baudrate immediately but stores the value and sets it at detection of the device. Same for setting to a wrong baudrate.

Koromix, 2019-11-29:
OK, what about the new-new-build? :p
https://koromix.dev/files/tytools/

luni64, 2019-11-29:
Did you change anything? It has the same build number as the one I tested

luni64, 2019-11-29:
Sorry, didn't refresh the browser....

luni64, 2019-11-29:
Perfect, switching baud rates works now. 

However, I get some additional blank lines in the output. (not a big deal but putty doesn't show them). Seems to work if I use the native Serial port instead Serial1 + cable.  This has nothing to do with the current changes, I observed them for quite while. 

[image]

Code: 
[image]

Koromix, 2019-11-29:
Hmmm, I may have an idea about that.
I just need to find my own USB-serial converter to do some tests (which may be the hardest part about it).

Koromix, 2019-12-01:
Does the new build (TyTools-0.8.11-16-gc874426) fix the extra-lines issue ?
I could not test it (still haven't found my converter), so it's based on a hunch.

luni64, 2019-12-01:
Yes, works perfectly. 
Thanks again, this tool gets better and better all the time!  :-)

Koromix, 2019-12-01:
Thanks for the test!

closed 2019-12-01 by Koromix