Koromix/tytools #45: Uploading of large files fails

luni64, 2019-11-30:
While testing your last version I stumbled over problems uploading large hex files. 

The following code generates a large flash image which can be uploaded with the **current** Teensy.exe but fails to upload with tyCommander. 
```c++
PROGMEM const char buffer[500 * 1024] = "Hello Buffer\0";
void setup()
{
  while (!Serial);
  Serial.println(buffer);
}
void loop() {}
```
Error message: 
```
Firmware 'C:\Users\lutz\source\Teensy\TestProject\.vsteensy\build\TestProject.hex' has too many segments
```
It works nicely if I reduce the buffer size to 400kB. Seems to be limited by 512kB instead of 2MB

Koromix, 2019-12-01:
Thanks for the catch! There were actually two small bugs for the T4: one in the IHEX parser, one in the "ELF upload" code.

As usual, a new build is available here:
https://koromix.dev/files/tytools/

luni64, 2019-12-01:
Works perfectly. Thanks for the quick fix.

closed 2019-12-01 by luni64

Defragster, 2019-12-01:
Good job posting the issue Luni!  Was glad to see you did that after it showed up!

Downloaded the 12/1 zip from link above and it worked on the one sketch I uploaded so far.

Hi Koromix!  Good work keeping TyCommander as the best thing about using Teensy - that PJRC doesn't make.

It is nice integrated into the IDE - even better with the FrankB CMDLine trigger incorporated in defragster\TSET that works well with SublimeText - Though Luni has done a great job with using TyComm with his VSCode variant as well.