Koromix/tytools #47: 'Send data ...' history include sent "Send File"

Defragster, 2019-12-02:
Would it be difficult to include in the Send history list 'Send File: xyz.txt'?

In testing I may send a file ... then the same file again … or alternate between multiple files.

It would be handy to have the History list include a "token" to resend the same file - without having to go through hitting the Send Drop down and Send File and select the file each time.

Koromix, 2019-12-02:
Done in the latest version, though I won't be able to publish builds until tonight!
Everything is done for 0.9.0 so that's what I will publish tonight.

closed 2019-12-02 by Koromix

Defragster, 2019-12-02:
Great! Looking forward to it.  Will be very handy for pushing out known large data stream to T4 for USB testing.

Just installed TD 1.49b1, used IDE once to test - then did TyCommander Integrate and moved to use in SublimeTest and it is working as expected - with the prior posted build I got.

I even tested that LARGE FLASH array alloc sketch @luni did and it worked