Koromix/tytools #50: Multiple Serial Support

luni64, 2020-02-22:
Teensy supports now multiple serial ports per device.  (Here the corresponding pull request.  https://github.com/PaulStoffregen/cores/pull/438).  This, of course is a great feature if you need to debug something which uses the normal Serial port for communication with the PC.  As can be expected, the current tytools don't support this yet. 

I tried it with the following sketch:
```c++
void setup()
{
  pinMode(LED_BUILTIN,OUTPUT);
}

void loop()
{
  Serial.printf("Serial %d\r\n", millis());
  SerialA.printf("SerialA %d\r\n", millis());   // Second Port

  digitalWriteFast(LED_BUILTIN,!digitalReadFast(LED_BUILTIN));
  delay(500);
}
```
Which works nicely. Both ports appear in Win10 and can be connected to with putty.  However tytools gets confused. It does show the new ports in the info screen as shown below:

[image]

COM11 was the original port, COM 29 and 30 are the two new ones. 

But, unfortunately, reprogramming doesn't work. It looks like it looses the Teensy after programming. Probably it doesn't expect changed Ports Numbers? 

Attached please find a HEX file with the sketch from above in case you want to give it a quick try. Would love if tytools could support this.

Defragster, 2020-02-24:
Paul posted that this will be in the next TD beta for #1.52!
Very cool to have a way to separate debug Spew perhaps from core 'Sketch I/O'

@Koromix - nice if you have the time to work with this.  Might help with Teensy 4 I/O speed two if the different 'ports' can be placed into different TyCommander windows.  Also might make it easier for future work to leave primary ports as it is and put VT100 support on other(s).

Koromix, 2020-03-01:
Hi,
I've started to work on this but it will take some time to do properly.
Still, here's what I've got so far after some hacking:
[image]

luni64, 2020-03-01:
Cool, looking forward to get this. Will make debugging MUCH nicer. 

I don't know if you noticed, meanwhile all ports can be used to reboot the teensy. The hex I posted above was done with an older version where only one port accepted reboot requests.

Koromix, 2020-03-01:
Yes this works already.
I need to clean up the changes needed (right now some parts are hacked in) and work on proper board settings in this case.
And I'd like to show in the GUI that the two Teensies are linked somehow.

luni64, 2020-03-01:
> And I'd like to show in the GUI that the two Teensies are linked somehow, with one of them being the "master" (ideally Serial and not SerialUSB1 / etc.).

Yes, while I was playing with Putty I already noticed that identifying Serials looking at the port numbers gets confusing quickly. A clear indication which port corresponds to which Serial would be helpful indeed.

Koromix, 2020-03-01:
I will work on this more later, but in the meantime you can find a build with early multi-serial support here: https://koromix.dev/files/tytools/

Defragster, 2020-03-02:
Nice work Koromix!

Defragster, 2020-03-03:
Just tested with current PJRC from github.

It shows three 'Teensy' from one Teensy 4.0!

PROBLEM - maybe this is known future work: The goal is each of the three should have their own 'Serial' output GUI window, currently all three map to the same GUI window.

luni64, 2020-03-03:
> The goal is each of the three should have their own 'Serial' output GUI window, currently all three map to the same GUI window.

Yes, will be useful sometimes.  Usually, if I have more than one port per Teensy (which needed USB Serial cables in the past...) I simply open a second window. Advantage is that you can see both terminals in parallel. 

It would be good to be able to have a color setting (say title bar, or some other visible part of the GUI) The setting should be stored based on Teensy ID + selected Port. This would make it easy to identify the port/terminal on a glace.

Defragster, 2020-03-03:
I just noticed that each of the 3 instances on the options tab do point to a unique TEMP file!
So the output is logged exclusively to a unique file for each.
There may be times when 'all on one would make sense'  - but my expected use for this was 'multimon' type use to see DEBUG SPEW and USEFUL key info on separate GUI pages.

Also hoping that will split the output load from the faster Teensy 4.0 in a way that total throughput between the 2 or 3 windows might exceed what one can do now.

Koromix, 2020-03-03:
I'm not sure I understand, and I can't retest it right now, but if you open two TyCommander windows and set each one to a different "Teensy port" you should be able to watch both outputs at the same time?

Koromix, 2020-03-03:
And you can use the compact window mode with Ctrl+H or "View > Compact Mode" to reduce used screen space.

Defragster, 2020-03-03:
Let me check - I didn't read the code - it is doing something unexpected …

Defragster, 2020-03-03:
MY Opps - YOU DID PERFECTLY right Koromix!  Thank You!

Works with sample code rewrite to work to actually show the desired unique output:
void loop() {
  if ( Serial.available() > 0 ) {
    Serial.print("Main Serial :: ");
    while ( Serial.available() > 0 )
      Serial.print( (char) Serial.read());
    Serial.println();
  }
  if ( SerialUSB1.available() > 0 ) {
    SerialUSB1.print("Serial USB1 :: ");
    while ( SerialUSB1.available() > 0 )
      SerialUSB1.print( (char) SerialUSB1.read());
    SerialUSB1.print("\n");
  }
  if ( SerialUSB2.available() > 0 ) {
    SerialUSB2.print("Serial USB2 :: ");
    while ( SerialUSB2.available() > 0 )
      SerialUSB2.print( (char) SerialUSB2.read());
    SerialUSB2.print("\n");
  }
}

Defragster, 2020-03-03:
Is each USB# associated handled independently - different processor core? - or common?

Koromix, 2020-03-03:
Serial reads are handled with two threads: one reads from all Teensies, the second one deals with the GUI.
The performance problems happen in the second one, and it cannot be changed because the Qt GUI is fundamentally single-threaded.

Defragster, 2020-03-03:
Good info.  Thanks for the support.  This would be awful without TyCommander!
My machine seems slow - time for reboot it seems.  IDE was no better, even Single Serial only.

Koromix, 2020-08-01:
I've made some progress on this; here is a new Windows build: https://koromix.dev/files/tytools/ (TyTools-0.9.1-10-g65d523b).

The label now contains the interface number (@2, @4), and each "board" should keep its own settings. They are also ordered deterministically in the main window, and a little + is shown for "secondary boards".

luni64, 2020-08-01:
Gave it a quick test. Runs much more stable now. I saw that you removed reboot and reset from the auxiliary ports which is good (less confusing). However, on first connection to a Teensy it still has all ports in the list where it asks you to select the board for uploading. If you choose the @2 port it can't upload of course. 
 
Not  a big thing but might confuse new users. 

[image]

Thanks a lot for this. Your TyTools are extremely useful productivity enhancers for me.

Defragster, 2020-08-01:
Nice add with the @2 - Will give it a look 

Expect the UI will work well.  Would it be easy to do a 'multi' select automatically when any of the exposed ports on a single Teensy are selected?  By excluding the 'non-primary' what happens if the 'Primary' is connected by an outside USB Serial connect? Another SerMon or PC application?

Koromix, 2020-08-01:
Yes, I was looking at that (the selection dialog). In the meantime, there's a new build with very minor GUI fixes, such as selecting the primary board by default after launch (it would select kinda randomly in the previous build).

Koromix, 2020-08-01:
> Nice add with the @2 - Will give it a look
>
> Expect the UI will work well. Would it be easy to do a 'multi' select automatically when any of the exposed ports on a single Teensy are selected? By [...]

Not sure I understand. If I select multiple "boards" automatically, you can't look at the serial output. Or maybe I don't understard?

Defragster, 2020-08-01:
Maybe I'm not following or explaining ... This is during 'Program' device selection with MultiSerial Teensy:

Selecting ANY of the USB ports is the same as selecting ALL of the ports for that device to program it.

If 2500450-Teensy has 2 or 3 USB 'ports' - when any of them is picked - can it Multi-Select All of them for clarity and avoid picking one and having it fail?

luni64, 2020-08-01:
Does the read '+' just marks the secondary ports or is it showing any other info? If possible It might be good to replace the '+" by "1" and "2". The @2 / @3 in the board name is good, but since one can rename it (and I do that often), you'd loose that info. Having it in the icon might be better...

Defragster, 2020-08-01:
Alternate ICON '1'/'2' makes sense for USB1 and USB2.
Here is a T_4.1 with triple Serial - looking and working right it seems:
[image]

Not sure why Reset & Bootloader active in GUI on secondary ports is bad?  Controlling that from whatever instance is in front  is handy.

Koromix, 2020-08-01:
It's not bad, and I've changed this, you can do it in the latest build (which also uses little numbers instead of the '+' in the red circle).

Upload remains blocked because it is more problematic; in TyCommander there's a notion of "association between a firmware and a board". This thing does not know how to deal with the fact that multiple boards can in fact be the same. For now, it's blocked for secondary boards.

In this build, the upload selection dialog will not allow you to select a secondary board.

New build: https://koromix.dev/files/tytools/ (TyTools-0.9.1-15-g87ed6a6)

Defragster, 2020-08-01:
Looks Good and it works.

Turned back on integration and dialog for upload seems fine.

Defragster, 2020-08-02:
There is a new build with notes linked here [https://github.com/Koromix/tytools/issues/59](url) { [https://koromix.dev/files/tytools/](url) |

It is updated to have a command line way to control Teensy Reboot - and  it has updated **and great looking Multi USB support** as far as tested here.  When this gets released I can update the [https://github.com/Defragster/Tset](url) code to use it for cases where the IDE does not have TyCommander integrated.  Selecting Teensy once for a sketch pushed it to bootloader for TeensyLoader AUTO program.

Using two T_4.1's alternating on from 2 to 3 USB's in unique sketches while the other running a Dual Serial sketch All seems well!

**Thanks again @Koromix**  - without TyCommander programming Teensy would only be half as awesome!

Koromix, 2020-08-03:
Thanks :) I've fixed the URLs in your comment.

Defragster, 2020-08-03:
URL's do work now - Thanks - I dropped them and ran without testing it seems :(

Defragster, 2020-08-07:
@Koromix Are you ready to make this the main release version?  It has been running well here - just sitting.  Supporting Multi USB is a useful feature not in the current release and with the UI improvements are a good step forward.  And having the --delegate works as desired and won't break anything when not used.

Then I can update TSET to expect it for cases where TyCommander isn't integrated at the time and --delegate takes out the manual bootloader trigger step where it otherwise blocks Teensy Loader.

Koromix, 2020-08-07:
Good idea :) I've made some tests on macOS, it seems to work well.

Uploading builds, which takes 1 hour on my ~16kB/s upload connection... Recently moved, and went from 50 MB/s (UP) to 16 kB/s (UP). That's painful ><

closed 2020-08-07 by Koromix

Defragster, 2020-08-07:
The working on MacOS sounds GOOD!

The downgraded upload speed does not sound good :(

I thankfully went from 7Mbps to ~40Mbps which was nice  ... 160kbps sounds very painful.

Koromix, 2020-08-07:
Release done!

Defragster, 2020-08-07:
Cool, Updated https://github.com/Defragster/Tset readme and the batch files to call with --delegate when not integrated.