Koromix/tytools #52: Teensy :: _teensy_model_identifier

Defragster, 2020-03-10:
I just discovered that the .ld files posted for Teensy include the _teensy_model_identifier used to identify what Teensy is connected?

For T_4.0: T:\arduino-1.8.12\hardware\teensy\avr\cores\teensy4\imxrt1062.ld
_teensy_model_identifier = 0x24;

For unreleased T_4.1: T:\arduino-1.8.12\hardware\teensy\avr\cores\teensy4\imxrt1062_t41.ld
_teensy_model_identifier = 0x25;

If TyTools could recognize { for Serial connect only until after release } it would be properly identified and handled for Serial Monitoring when the T_4.1 releases.

KurtE, 2020-03-17:
@Defragster @Koromix - I think I have a version that will probably work when T4.1 is released. 

It is based off of your multi stream branch:
https://github.com/KurtE/tytools/tree/T4_1_MS

I could do a PR, but may want to wait until we can fully verify it works when the T4.1 is released

Koromix, 2020-06-07:
Merged it, thanks!
Windows builds for master and multiple_serials branches can be found here: https://koromix.dev/files/tytools/

Koromix, 2020-06-07:
Windows builds for master and multiple_serials branches can be found here: https://koromix.dev/files/tytools/

closed 2020-06-07 by Koromix