Koromix/tytools #53: MultiSerial uploads to ALL instances

Defragster, 2020-03-18:
Just found on a T_4.0 with Triple_Serial calling TyCommander when a single T_4.0 is online ( already running Triple_Serial ).
CMD_LINE:: "%TyTools%\TyCommanderC.exe" upload --autostart --wait  "%temp1%\%sketchname%.%model%.hex"

TyCommand takes the request it seems to program all three of those Teensy interfaces - on the single Teensy.

> when the Parallel tasks upload is >1 it tried two attemtps to the same device and fails with Flash left EMPTY
> When Parallel is ==1 it then programs the same device 3 times in turn.

Defragster, 2020-06-07:
I don't see an issue here with current test release today on T_4.1.

Koromix, 2020-07-30:
I'm closing this to clean up a little but looking a bit deepter into how command-line mode behaves with multi-serial stuff is on my TODO list for when I can work on the multiple_serials branch again.

closed 2020-07-30 by Koromix