Koromix/tytools #55: Send window filled with previous input

paynterf, 2020-06-03:
The 'send' window at the bottom of the Serial window gets filled with the previous input as soon as any character is entered.  For instance, entering the character '-' causes the send window to show '-200' as '200' was the last entered string.  These characters are overwritten one at a time by the new input, but if the new send string is shorter than the previous ones, the remaining characters from the previous string are sent with the new ones.  In the example where the previous input was '200', then an input of '-10' becomes '-100', and the '-100' is what gets sent.

[image]

In the above image, I typed '-' into the 'Send' box. 

Frank

Defragster, 2020-06-03:
Yes, that is a history 'feature' for auto complete

Mixed results nice when long string to repeat, have to hit 'Delete' when not desired.

paynterf, 2020-06-03:
Well, at least now I know I'm not entirely crazy! ;-)

Defragster, 2020-06-03:
That is promising to know @paynterf  ... the whole world is therefore "not entirely crazy! ;-)"

paynterf, 2020-06-06:
I'm re-opening this issue, as it has become a real user-interface problem.  I have a rotary scan table program that accepts multiple parameter inputs from the user - rpm, initial position, start/stop scan positions, and number of scan steps. Each parameter has a default value, so the user could just send ENTER a bunch of times and get the default behavior, but that is rare.  What is more common is the user will accept the defaults for some parameter, and input values for some others. 

My current testing setup has me entering the following sequence
ENTER (accept default RPM value)
Q  ENTER (accept current table position as the zero position
-30 ENTER (set scan start position)
30 ENTER (set scan stop position)
6 ENTER (set number of steps)

What actually happens when I enter these number in this order is a scan start position of -30 as desired, but a scan stop position of 300 (almost a full revolution), and 60 steps instead of 6, due to a zero being added to the input value each time after the first numeric input (in this case '-30').

I realize I can backspace out the added zero each time, but this is counter-intuitive in the extreme, and having my table rotate 330 degrees (-30 to +300) instead of 60 (-30 to 30) is ... "exciting" to say the least, as I'm using a NEMA 17 stepper motor with LOTS of torque.  And, having the scan go through 60 steps instead of 6 is a bit disconcerting as well.

Please remove this 'feature', or at least give us mere mortals a way to turn it off ;-).

TIA,

Frank

Defragster, 2020-06-06:
As noted above 'Mixed results' - some half the time it is not helpful.

Perhaps under 'Send' dropdown a switch for 'History On/Off' like for CR and LF?

paynterf, 2020-06-06:
That would work fine for me

Frank

Koromix, 2020-06-07:
Unless someone changes my mind, the current behavior seems completely undesirable to me.
I've pushed a commit to disable it entirely, with no possibility to reenable it.

I'll publish Windows dev builds soon (and give the link when available) without this behavior.

Thanks for the report!

closed 2020-06-07 by Koromix

luni64, 2020-06-07:
I agree on the autocomplete feature, this is a PITA. However, scrolling through the history with up/down arrows is extremly useful. Hope that stays?

Koromix, 2020-06-07:
This stays! Only the default completion goes away.

Koromix, 2020-06-07:
Windows builds for master and multiple_serials branches can be found here: https://koromix.dev/files/tytools/

paynterf, 2020-06-07:
Uh, for us dummies, what's the difference between the 'master' and 'multiple-serials' branches?

TIA,

Frank

Koromix, 2020-06-07:
You can read the first post in the corresponding issue for more information about the multiple_serials branch: https://github.com/Koromix/tytools/issues/50

Eventually this will turn into a new feature for the next release. If you were using a release build (made from the master branch), you can ignore it and take the master branch build.

paynterf, 2020-06-07:
Thanks!

Frank

commit 058715ba (2020-06-07): Disable default completer for serial input (fixes #55)