Koromix/tytools #56: Generic Reboot:

Defragster, 2020-06-26:
Wondering if there might be a way to expose this value of '1200' versus Teensy's reboot value for a generic device? And then do that toggle for reset?

This line below from IDE build:  Forcing reset using 1200bps open/close on port COM36
------> IDE Board Info
BN: Adafruit Feather Bluefruit Sense
VID: 239A
PID: 8087
SN: DDF154F73969DFDD

------> TyCommander info:
TyC_ID: DDF154F73969DFDD-Adafruit_LLC
TyC_ Desc: Feather nRF52840 Sense
TyC_ Model: Generic

------> IDE console output
Sketch uses 38076 bytes (4%) of program storage space. Maximum is 815104 bytes.
Global variables use 6764 bytes (2%) of dynamic memory, leaving 230804 bytes for local variables. Maximum is 237568 bytes.
Forcing reset using 1200bps open/close on port COM36

Defragster, 2020-06-27:
I'm also using the adafruit CLUE that uses the same MCU - may have a diff PID/VID if needed I can post - but expect it uses the same trigger for DFU upload

Koromix, 2020-07-30:
I may expose this, but I'm not sure, and probably not in the GUI. There's a (undocumented) mechanism in libty to patch VID:PID pairs and rename device types using an INI file. Which I barely remember anything about myself, I just had to dig it out of the source code.

Maybe I can augment it a little for this kind of thing.

Koromix, 2020-08-02:
Mmmmhh, on second thought, I'm not sure I want to go through with this.

What I'm willing to do is support reset for a bunch of non-Teensy boards directly in the code, if you give me the VID:PID pair and the reset baudrate. And another bit of information: does the baudrate cause a reboot (go to bootloader and wait there) or a reset (restart the board and run the sketch)?

But I'm probably not going to make something that can be customized.

Defragster, 2020-08-02:
I can't say for sure - it probably needs to trigger the reboot baud sequence and then 'go offline' until the build completes.

It uses a DFU programming process.  And it doesn't wait to upload like TeensyLoader in AUTO mode.

It is probably best to ignore this issue. And require the current manual TyCommander 'Serial Offline' then trigger the IDE build and DFU upload - then wait for it to complete and then reconnect the USB Serial.

Defragster, 2020-08-02:
Going to close - to odd - likely require custom hacks - to be worth the trouble.

closed 2020-08-02 by Defragster