Koromix/tytools #58: Dual Serial - How to select the displayed serial port

luni64, 2020-07-21:
WIN10. I can upload a dual serial firmware and the information screen shows both ports. I want to connect the serial monitor to one of the two ports but how can I select which one to connect? It looks like it always connects to the first one in the list. Am I missing something?

Defragster, 2020-07-21:
The Information tab shows the  USB COM port for each uniquely for each entry on the left pane of devices.

Select one for non-SerMon use and click the 'Serial' button to discconect it from TyComm and then the PC can connect to that one using the : "\\.\COM29" type number.

One will be Serial and the other the SerialUSB1 (? - always forget the name).  If they are output backwards - go to prior step and swap the selection.

Once associated and viewed the association seems to stick when that Teensy is reprogrammed.

Hoping that answers the question?

luni64, 2020-07-21:
Thanks Tim. Tried it but it doesn't seem to run stably. Looks like I can only change the used port once after restarting TyCommander. Further changes only work very seldom for me. 

I was looking forward to have a second port for debug messages, but it needs to work reliably to be really useful. 

Tried it with putty, both ports print their string perfectly. So must be something on the TyCommander side. 

Test sketch:
```c++
void setup()
{
}

void loop()
{
  Serial.println("Serial");
  SerialUSB1.println("SerialUSB1");
  delay(100);
}
```

Additionally I sometimes get weird error messages: 

[image]

They seem to appear when I reset the board while the serial monitor is active.

Defragster, 2020-07-21:
The PUBLISHED update to TyCommander doesn't support Dual Serial AFAIK?  It should be showing "BOTH" entries on the left pane.

The DUAL Serial link to windows.exe/ZIP is on a recent 'closed issue' that led to the update?  That is the one I am using.

Defragster, 2020-07-21:
Looks like it may be here - from Koromix Post.

Merged it, thanks!
Windows builds for master and multiple_serials branches can be found here: https://koromix.dev/files/tytools/

luni64, 2020-07-21:
Using 0.9.1 and it shows both Serials in the info tab. Upload also works so seems to be correct. Anyway, I'll try your linked version  (tomorrow). Hopefully this works better. 

[image]

Defragster, 2020-07-21:
Each should be uniquely populated in left pane - that 'release' version could be updated as the MULTI serial works fine for my use.
[image]

luni64, 2020-07-21:
Ah, that looks better. I'll try it asap

Koromix, 2020-07-21:
Yeah, the plan was to improve the multiple_serials branch a bit on two points: (1) the code is a bit hacky and fragile, (2) I want to tweak the left pane to show that both entries belong together (same Teensy) somehow.
As you can see progress on TyTools is now extremely slow. Hopefully I can find some time to work on this soon, and merge all of this into the master branch.

Koromix, 2020-07-21:
On a related note, thanks to your screenshot I can see that the HiDPI support is not perfect. The font in the left panel is a bit too big :D That's probably something I can work on for 0.9.2

(I don't have a HiDPI screen so I don't notice this kind of problem)

Defragster, 2020-07-21:
HiDPI - like 3840x2160?  Have it running at 125% scale to make it human readable on 28" screen.
I do notice that the Upload 'choose Teensy' dialog text in list runs off the right on this screen - if I move to 2nd screen it is readable to see which Teensy to upload to. Here it is twice on main screen - the second time I resized it wider.
[image]

> One handy thing on Multi Serial would be showing the COM# there.  Doing GDB or other requires knowing the Com# and disabling the Serail on the 'right' one as noted by @luni64  - don't know if you can tell from the descriptor which is Serial and which SerialUSB[1,2] - could save multiple clicks into the information tab 

Good luck finding time, and resolving UI issues to your satisfaction.

The only odd thing I saw with Multi was at one point on upload it would hit setup print a line there and then restart and print that again and then continue.  Not sure if that was GSB at the time - or if TyComm was hitting a reset for each instance after upload to a single Serial# Teensy?

luni64, 2020-07-22:
Tried the linked version. Works good so far. Thanks a lot this is really very useful. 

Small findings: 
- I can rename the two (virtual) boards to e.g. Serial & usbSerial which is good. Unfortunately it looses the names after a restart of TyCommander and renames both with the first name. 
- It seems to ignore the option "Enable serial connection by default" (it always connects)

luni64, 2020-07-30:
One additional comment: 
It would be perfect if you could display the "type" of the Serial the port connects to. I.e, whether it is Serial, SerialUSB1 or SerialUSB2.  Looking at usb_desc.h and the Windows device manager the Serial, SerialUSB1 and SerialUSB2 correspond to the usb interfaces 0, 2 and 4 respectively.

Koromix, 2020-08-01:
I've made some progress on this (and a new build). I'm closing this bug because it is kind of redundant with the previous one: #50 
Go to #50 for the new build.

closed 2020-08-01 by Koromix