Koromix/tytools #59: TyCommanderC.exe trigger bootloader from -multi select list

Defragster, 2020-07-29:
Would it be easy and sensible to have the UPLOAD behavior presenting the 'choose board' dialog trigger a Reboot?

In the case of  platform.txt it uses :: upload --autostart --wait --multi

Having the ability to command line present the list and then trigger 'Bootloader' would in the case where TyCommander is not integrated but using the TSET command line build from editor would take the Teensy offline and trigger the TeensyLoader to Upload the Teensy before it returned.  Currently I have to click the GUI 'Bootloader' button - or the Teensy Button to disconnect TyCommander and allow TeensyLoader USB access to the desired Teensy.

This comes up during Beta releases of TeensyDuino when avoiding the IDE GUI is okay - but need to see TeensyLoader do the upload without having to so Integrate.

Using perhaps a variation of this that currently results in this with no prompting:
T:\TyComm>TyCommanderC.exe reboot --autostart --wait --multi
       reboot@7819600-Teensy  Rebooting board '7819600-Teensy' (Teensy 4.1)
       reboot@7819600-Teensy  Board is already in bootloader mode
       reboot@1245170-Teensy  Rebooting board '1245170-Teensy' (Teensy 3.2)
       reboot@1245170-Teensy  Board is already in bootloader mode

Koromix, 2020-07-30:
Right now, how it works is that the TyCommander instance triggers the selection dialog only for upload.

Two things can be done:
- Support "--select" without any option t force a selection dialog to always show up,
- Support HEX filename for reboot command and use it the same way we do for upload: find an associated Teensy, show a dialog if we can't find any.

Defragster, 2020-08-01:
This is an odd case where TyCommand not integrated - and TeensyLoader owns programming.

Without integration the 'command line IDE build' as used doesn't know what  Teensy to take offline or pass to TeensyLoader as TyComm owns that USB port.

This just needs to offer an easy way to have TyComm drop the port and trigger TeensyLoader to program. If TyComm tried to program while TeensyLoader active it hangs TLoader.

This isn't critical - in current use the IDE build passes the HEX to TLoader and then I have to push button or hit 'Bootloader' in TyComm to get the TLoader to program.  If this change could present a dialog it would make it easier.

Koromix, 2020-08-01:
While I'm willing to improve this use case, this begs the question: is there a specific reason you prefer the official loader? Is it because of #41?

Defragster, 2020-08-01:
I really did this on HardWare Beta before TyComm was able to be updated to support new devices you didn't have.

One reason is when I'm doing Beta builds of TeensyDuino I do it to help test PJRC code - even though not directly in the IDE - at least it confirms that works.  Using my TSET for command line build it uses the installed IDE and TeensyDuino so that is tested too.  But I can pop into the IDE as well for stupid IDE SerMon.

Another is that only the TeensyLoader will provide Firmware Upgrades to the Teensy when they happen - that sort of ties in with the item above to make sure when they are delivered I get them.

Also indeed #41 - it would be nice if the RTC set code mechanism was exposed by PJRC.

Koromix, 2020-08-02:
I've added a semi-hidden option to handle this situation.

You will need to edit platform.txt yourself (after integration), and add --delegate to the list of options:  

* Before: ```tools.teensyloader.upload.pattern="{cmd.path}" upload --autostart --wait --multi {upload.verbose} "{build.path}/{build.project_name}.{build.board}.hex"`
* After: ```tools.teensyloader.upload.pattern="{cmd.path}" upload --autostart --wait --multi --delegate {upload.verbose} "{build.path}/{build.project_name}.{build.board}.hex"`

With this option, the upload task will let the Teensy Loader do its stuff. TyCommander won't talk to the loader; you first need to point the Teensy Loader (manually or by adding a command to platform.txt) to the correct HEX file. TyCommander will assume you have done so, and once the board becomes available again it will assume the HEX file has been uploaded by the loader.

Build is here: https://koromix.dev/files/tytools/ (TyTools-0.9.1-18-gd7e26e9)

Defragster, 2020-08-02:
**That works PERFECTLY!**  
> TyCommander not integrated to IDE - no change to platform.txt - just an edit to the batch file
> Start Build and TyComm select dialog appears - with foreground focus - the chosen Teensy goes OFFLINE and TeensyLoader programs the Teensy that comes back online in TyComm.
>> Do a REBUILD of same sketch and - without prompting the Same Teensy get programmed and restart.
> Multiple Teensy online - with two skeches to two Teensy and all works well!

The 'upload.pattern' code runs from SublimeText Editor batch file - it doesn't use any changes to platform.txt.
The Batch file uses the IF line when TyComm is Integrated and USES the ELSE when not:

    if EXIST "%temp1%\%sketchname%.%model%.hex" (
      "%TyTools%\TyCommanderC.exe" upload --autostart --wait  "%temp1%\%sketchname%.%model%.hex" ) else ( 
      "%TyTools%\TyCommanderC.exe" upload --autostart --wait **--delegate** "%temp1%\%sketchname%.hex" )

NOTE: That temp build does not have the RED around the Multi Serial# and the number only half shows in white on the black:
[image]

Defragster, 2020-08-02:
Okay - not quite perfect?  I did a DROP association and upload after prompt that shows running but TyCommander is confused?

Program worked - but had to close TLoader - and hit Button to take both T_4.1's offline - then hit GUI Reset ?

Did a drop Assoc on the one running 'SerialUSB' with 3 USB
Then did Upload to the one running 'Entropy' with only 2 USB - it went from 2 to 3 and then both Teensy 
> Just repeated putting the 2 USB 'Entropy' on both 

BTW - in this case the Association labels are not updated?

Defragster, 2020-08-02:
Perhaps with '--delegate' it would be okay to always prompt for device?

On the 3 USB Serial sketch it takes input and prints on ALL three - getting 'busy on task? Have to kill TyComm and background processes?

Listing USB host controllers and root hubs
Found root USB hub 'USB\ROOT_HUB30\4&362721B3&0&0' with ID 0
Examining device node 'BTHENUM\{00001101-0000-1000-8000-00805F9B34FB}_VID&00000000_PID&0000\7&13851A9D&1&B8AD3E90287B_C00000000'
Unknown device type for 'BTHENUM\{00001101-0000-1000-8000-00805F9B34FB}_VID&00000000_PID&0000\7&13851A9D&1&B8AD3E90287B_C00000000'
Examining device node 'USB\VID_16C0&PID_048B&MI_00\7&39937848&0&0000'
Found port number of 'USB\VID_16C0&PID_048B\7684140': 2
Found port number of 'USB\VID_045B&PID_0209\5&38BA1A52&0&5': 5
Found controller ID for 'USB\ROOT_HUB30\4&362721B3&0&0': 1
Add serial device 'USB\VID_16C0&PID_048B&MI_00\7&39937848&0&0000' on iface 0
  - USB VID/PID = 16c0:048b, USB location = usb-1-5-2
  - USB manufacturer = Teensyduino, product = Dual Serial, S/N = 7684140
Examining device node 'BTHENUM\{00001101-0000-1000-8000-00805F9B34FB}_LOCALMFG&0000\7&13851A9D&1&000000000000_00000000'
Unknown device type for 'BTHENUM\{00001101-0000-1000-8000-00805F9B34FB}_LOCALMFG&0000\7&13851A9D&1&000000000000_00000000'
Examining device node 'USB\VID_16C0&PID_048B&MI_02\7&39937848&0&0002'
Found port number of 'USB\VID_16C0&PID_048B\7684140': 2
Found port number of 'USB\VID_045B&PID_0209\5&38BA1A52&0&5': 5
Found controller ID for 'USB\ROOT_HUB30\4&362721B3&0&0': 1
Add serial device 'USB\VID_16C0&PID_048B&MI_02\7&39937848&0&0002' on iface 2
  - USB VID/PID = 16c0:048b, USB location = usb-1-5-2
  - USB manufacturer = Teensyduino, product = Dual Serial, S/N = 7684140
Examining device node 'USB\VID_16C0&PID_048B&MI_02\7&140786D7&0&0002'
Found port number of 'USB\VID_16C0&PID_048B\7819600': 4
Found port number of 'USB\VID_045B&PID_0209\5&38BA1A52&0&5': 5
Found controller ID for 'USB\ROOT_HUB30\4&362721B3&0&0': 1
Add serial device 'USB\VID_16C0&PID_048B&MI_02\7&140786D7&0&0002' on iface 2
  - USB VID/PID = 16c0:048b, USB location = usb-1-5-4
  - USB manufacturer = Teensyduino, product = Dual Serial, S/N = 7819600
Examining device node 'USB\VID_16C0&PID_048B&MI_00\7&140786D7&0&0000'
Found port number of 'USB\VID_16C0&PID_048B\7819600': 4
Found port number of 'USB\VID_045B&PID_0209\5&38BA1A52&0&5': 5
Found controller ID for 'USB\ROOT_HUB30\4&362721B3&0&0': 1
Add serial device 'USB\VID_16C0&PID_048B&MI_00\7&140786D7&0&0000' on iface 0
  - USB VID/PID = 16c0:048b, USB location = usb-1-5-4
  - USB manufacturer = Teensyduino, product = Dual Serial, S/N = 7819600
Ignoring duplicate HID collection device 'HID\VID_046D&PID_C534&MI_01&COL03\9&3A725E9F&0&0002'
Examining device node 'USB\VID_046D&PID_C534&MI_01\8&27DFA116&0&0001'
Device 'USB\VID_046D&PID_C534&MI_01\8&27DFA116&0&0001' has no 'PortName' registry property
Examining device node 'BTHENUM\{00001124-0000-1000-8000-00805F9B34FB}_VID&0002046D_PID&B33D\7&2A21DF93&0&34885D78E314_C00000000'
Unknown device type for 'BTHENUM\{00001124-0000-1000-8000-00805F9B34FB}_VID&0002046D_PID&B33D\7&2A21DF93&0&34885D78E314_C00000000'
Ignoring duplicate HID collection device 'HID\VID_1B80&PID_B406&MI_01&COL03\9&17206C4E&0&0002'
Examining device node 'USB\VID_0764&PID_0501\6&40DFE5C&0&3'
Device 'USB\VID_0764&PID_0501\6&40DFE5C&0&3' has no 'PortName' registry property
Examining device node 'HID\*UPDDHID\1&306A472D&0&0000'
Cannot get parent of device 'HTREE\ROOT\0': 0xd
Ignoring duplicate HID collection device 'HID\VID_046D&PID_C534&MI_01&COL04\9&3A725E9F&0&0003'
Examining device node 'ROOT\*UPDDHID\0000'
Unknown device type for 'ROOT\*UPDDHID\0000'
Examining device node 'ROOT\*UPDDHID\0001'
Unknown device type for 'ROOT\*UPDDHID\0001'
Examining device node 'ROOT\*UPDDHID\0002'
Unknown device type for 'ROOT\*UPDDHID\0002'
Ignoring duplicate HID collection device 'HID\VID_1B80&PID_B406&MI_01&COL04\9&17206C4E&0&0003'
Examining device node 'HID\*UPDDHID\1&7892DCB&0&0000'
Cannot get parent of device 'HTREE\ROOT\0': 0xd
Ignoring duplicate HID collection device 'HID\VID_046D&PID_C534&MI_01&COL05\9&3A725E9F&0&0004'
Examining device node 'HID\VID_1B80&PID_B406&MI_02\9&CB6D174&0&0000'
Found port number of 'USB\VID_1B80&PID_B406\7&2F0A1E4C&0&1': 1
Found port number of 'USB\VID_1A40&PID_0101\6&40DFE5C&0&4': 4
Found port number of 'USB\VID_1A40&PID_0101\5&38BA1A52&0&8': 8
Found controller ID for 'USB\ROOT_HUB30\4&362721B3&0&0': 1
Examining device node 'USB\VID_1B80&PID_B406&MI_00\8&31E2AF39&0&0000'
Device 'USB\VID_1B80&PID_B406&MI_00\8&31E2AF39&0&0000' has no 'PortName' registry property
Examining device node 'BTHENUM\{00001124-0000-1000-8000-00805F9B34FB}_VID&0002045E_PID&0762\7&2A21DF93&0&002248872E59_C00000000'
Unknown device type for 'BTHENUM\{00001124-0000-1000-8000-00805F9B34FB}_VID&0002045E_PID&0762\7&2A21DF93&0&002248872E59_C00000000'
Examining device node 'BTHENUM\{00001124-0000-1000-8000-00805F9B34FB}_VID&000205AC_PID&022C\7&2A21DF93&0&CCC50A28857C_C00000000'
Unknown device type for 'BTHENUM\{00001124-0000-1000-8000-00805F9B34FB}_VID&000205AC_PID&022C\7&2A21DF93&0&CCC50A28857C_C00000000'
[... 35 lines cut ...]
Add serial device 'USB\VID_16C0&PID_048C&MI_00\7&185EE2AC&0&0000' on iface 0
  - USB VID/PID = 16c0:048c, USB location = usb-1-5-2
  - USB manufacturer = Teensyduino, product = Triple Serial, S/N = 7684140
Identified 'Teensy 4.1' with bcdDevice value 0x280
Received arrival notification for device 'USB\VID_16C0&PID_048C&MI_02\7&185EE2AC&0&0002'
[... cut ...]
Found controller ID for 'USB\ROOT_HUB30\4&362721B3&0&0': 1
Add serial device 'USB\VID_16C0&PID_048C&MI_04\7&185EE2AC&0&0004' on iface 4
  - USB VID/PID = 16c0:048c, USB location = usb-1-5-2
  - USB manufacturer = Teensyduino, product = Triple Serial, S/N = 7684140
Identified 'Teensy 4.1' with bcdDevice value 0x280
[upload@7684140-Teensy] Reboot didn't work, press button manually
Board '7684140-Teensy@2' is busy on task 'send@7684140-Teensy@2'
Board '7684140-Teensy@2' is busy on task 'send@7684140-Teensy@2'

Defragster, 2020-08-02:
It works fine for multiple uploads - until I try Drop Association to put the code on another Teensy.

Koromix, 2020-08-02:
It decides to prompt if it can't find a Teensy associated with the firmware path. It does not care about the model at this stage.

When you integrate TyCommander with Arduino, the path of the firmware is changed a bit to always include the board model. So if you change from Teensy 4.0 to Teensy 4.1 (for example) for the same sketch, the HEX file will have a different path. Without integration, this is not the case.

You could do the same change/hack in platform.txt to fix this problem, without doing the integration stuff.

Koromix, 2020-08-02:
The "red circle bug" in the build was caused by an older version of Qt. I made this build on my old MacBook, and it did not want to rebuild Qt because it is so sloooowww ><.

I've made a new build, it should work correctly. Also, this new build has a fix for Wndows 7, on which dual/triple serial Teensies would not work correctly in TyCommander.

Defragster, 2020-08-02:
Indeed, the batch file code posted above differentiates based on seeing if the model is in the HEX name

if EXIST "%temp1%\%sketchname%.%model%.hex" (
  "%TyTools%\TyCommanderC.exe" upload --autostart --wait  "%temp1%\%sketchname%.%model%.hex" ) else ( 
  "%TyTools%\TyCommanderC.exe" upload --autostart --wait --delegate "%temp1%\%sketchname%.hex" )

When integrated it runs the Proper TyComm first line, without that it then uses the default sketch.HEX.

Frank Boesing crafted the Windows Command Line code that runs the IDE build - and I wrote a further batch file that prompts for all the board settings so the resulting batch file was easy to create or change for each sketch folder.

It prompts user for and then sets these values:
set model=teensy41
set speed=600
set opt=o2std
set usb=serial2

So the IDE GUI never has to be touched . The IDE builds { with or without TyComm Integration } and then works afterwards.

It is very handy and the SublimeText editor I use can trigger the needed batch file with its 'Build' system and capture the output very nicely, but the same also works from the Command line - or even Windows File Explorer to trigger a build.

It works now - as long as the Teensy doesn't change or it isn't needed on multiple Teensy or a different Teensy.

But in the case of  **TyCommanderC --delegate** it would work well if the test for " find a Teensy associated with the firmware path" was removed - and it always Prompted.

I need to restart my computer for an update before I can try the new code - there is an Update pending and also I can't overwrite the TyComm files for some reason?  I don't see where - but some process has the files open and locked by the system.

Defragster, 2020-08-02:
**Looks very good.**
[image]

I just did a Drop Association three times to change the code on the One T_4.1 and all seems well going from 2 to 3 USB and back with alternate sketch using the **--delegate** in my batch file with no changes to IDE { integration or platform.txt edits }

The file name of the Sketch is properly tracking now which may have cleared up the issue I was seeing.

Koromix, 2020-08-07:
Seems to work, don't hesitate to comment again or open a new issue if not :)

closed 2020-08-07 by Koromix

Defragster, 2020-08-07:
Indeed - working well as tested.  Thanks.  Went back to Integrated TyComm and no issues there.  I never tried hacking platform.txt - but for use in Windows TSET batch file where it does or does not see the .model.HEX file I was able to detect when to trigger the --delegate to get TeensyLoader to work as needed.

Defragster, 2020-11-06:
@Koromix - just showed --delegate to @luni64 and it works well for  VisualTeensy as well it seems - to avoid conflict depending on the level of TyCommander integration