Koromix/tytools #60: Trouble building tycmd on macOS Catalina

blackketter, 2020-08-23:
I'm trying to build tycmd on macOS Catalina.  I have qt 5.15.0 installed via brew.  When I try run cmake per the README, I get this:

```
CMake Error at cmake/FindEasyQt5.cmake:50 (find_package):
  Could not find a package configuration file provided by "Qt5" with any of
  the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  src/tycommander/CMakeLists.txt:11 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/dean/src/tytools/build/darwin/CMakeFiles/CMakeOutput.log".
```

I actually am not trying to build the GUI apps yet, just tycmd, but it seems that cmake requires qt anyway.  Thanks in advance.

Koromix, 2020-08-23:
Hi,

Yes, I probably need to improve (or at least document) the CMake Qt handling a bit.
If you only want tycmd, you don't need Qt, but you need to tell CMake that.

This should work (in addition to your own flags and cmake options, if any) ; `cmake -DCONFIG_TYCOMMANDER_BUILD=0 -DCONFIG_TYUPDATER_BUILD=0 ../..`.

blackketter, 2020-08-23:
Thanks!  That did the trick.  Out of curiousity how do I build with the installation of qt that I have?  Or do I need a different installation?

Koromix, 2020-08-24:
I just tested it, and homebrew does not create standard links in standard directories (`/usr/local/include`, etc.) for Qt5.

The directory can be explicitely set with the `Qt5_DIR` parameter, but to make hings easier I've added the homebrew Qt5 path to the CMake script. So if you pull the latest changes (aeadbcef305ef3c7cb76af02514c02cc2432ddb4), it should work directly.

For Windows and macOS, I build TyTools with a static Qt distribution to get standalone executables. The homebrew version uses shared libraries, which means you'll have to distribute the Qt5 dylibs (and DLLs on Windows) if you package TyTools built with homebrew Qt5.

If you want to make static builds like I do, you can follow the instructions in [lib/qt5/README.md](lib/qt5/README.md). This will make a static Qt5 build. Because Digia likes to break static builds with each version, if you do so, you need to use the exact Qt versions specified in the README (Qt 5.9.9 on macOS)... (thanks Digia).

blackketter, 2020-08-24:
Fantastic, that worked great! (Once I figured out I needed to delete my build directory…)

The only thing that would make it easier would be to have a formula for homebrew tytools directly.  But I suppose that’s a separate feature request.  :)

Thanks again!

Koromix, 2020-08-25:
You're welcome!

closed 2020-08-25 by Koromix