Koromix/tytools #61: Firmware too big (max 1048576 bytes) in

MikeMaurice, 2020-09-02:
It appears that this may be an issue with Teensy 4.1. 
Do I gather that 4.x is not supported. I have tytools installed with vsCode. But, cannot get the hex files to upload.

Defragster, 2020-09-02:
Both Teensy 4.0 and Teensy 4.1 are fully supported in the latest and recent TyCommander releases during or after they are in Beta into Production.

Forum shows VsCode is in use ??? - would suggest that the resulting build is bad from that environment as prepared.

Koromix, 2020-09-02:
You might be using an old release, probably version 0.8.10. Use the About dialog to check.
The latest version is here: https://github.com/Koromix/tytools/releases/tag/v0.9.2

MikeMaurice, 2020-09-02:
You are correct. I somehow got the 8.10 version. Will try the 9.2 and report it here.

MikeMaurice, 2020-09-02:
Here is the result: fact #1, the upload succeeds, even though there is an I/O error. The newProject is just the blink the led and I proved that it is replacing the previous hex code which is completely different and loaded by Teensy Loader. Maybe you can make something out of the error msg.

Waiting for user selection
       upload@7680710-Teensy  Uploading to board '7680710-Teensy' (Teensy 4.1)
       upload@7680710-Teensy  Triggering board reboot
       upload@7680710-Teensy  Firmware: newProject.hex
       upload@7680710-Teensy  Flash usage: 14 kiB (0.2%)
       upload@7680710-Teensy  Uploading...
       upload@7680710-Teensy  Sending reset command
       upload@7680710-Teensy  I/O error while writing to '\\.\HID#VID_16C0&PID_0478#7&4D0CBA6&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}'
make.exe: *** [makefile:168: uploadTy] Error 1
The terminal process "C:\Windows\system32\cmd.exe /d /c L:/MyDownloads/VisualTeensy_v0_9_8_2/make.exe uploadTy -j -Otarget" terminated with exit code: 2.

MikeMaurice, 2020-09-02:
By the way, I am using Windows 7, SP2.

Defragster, 2020-09-02:
Mike - sorry I only inferred you might need ' latest and recent TyCommander  ' ... glad you got updated.

As Far as the Error @Koromix of course will know - but that is after 'Sending reset' so the programming should be done.

@Koromix : I've noticed that after programming TyCommander 'goes red' offline a bit after programming at time and I've been ignoring it - perhaps this is related - but in ignoring it I've not captured the output.  I just did that monitoring output and all went well uploading a triple Serial sketch:

[upload@6362860-Teensy] Uploading to board '6362860-Teensy' (Teensy 4.0)
[upload@6362860-Teensy] Triggering board reboot
Received removal notification for device 'USB\VID_16C0&PID_048C&MI_00\7&2B03508F&1&0000'
Received removal notification for device 'USB\VID_16C0&PID_048C&MI_02\7&2B03508F&1&0002'
Received removal notification for device 'USB\VID_16C0&PID_048C&MI_04\7&2B03508F&1&0004'
Received removal notification for device 'USB\VID_16C0&PID_048C\6362860'
Remove device 'USB\VID_16C0&PID_048C\6362860'
Remove device 'USB\VID_16C0&PID_048C\6362860'
Remove device 'USB\VID_16C0&PID_048C\6362860'
Received arrival notification for device 'USB\VID_16C0&PID_0478\0009B57E'
Examining device node 'USB\VID_16C0&PID_0478\0009B57E'
Device 'USB\VID_16C0&PID_0478\0009B57E' has no 'PortName' registry property
Received arrival notification for device 'HID\VID_16C0&PID_0478\7&113038DD&0&0000'
Examining device node 'HID\VID_16C0&PID_0478\7&113038DD&0&0000'
Found port number of 'USB\VID_16C0&PID_0478\0009B57E': 3
Found port number of 'USB\VID_045B&PID_0209\5&38BA1A52&0&5': 5
Found controller ID for 'USB\ROOT_HUB30\4&362721B3&0&0': 1
Add HID device 'USB\VID_16C0&PID_0478\0009B57E' on iface 0
  - USB VID/PID = 16c0:0478, USB location = usb-1-5-3
  - USB manufacturer = (none), product = (none), S/N = 0009B57E
  - HID usage page = 0xff9c, HID usage = 0x24
Identified 'Teensy 4.0' with usage value 0x24
[upload@6362860-Teensy] Firmware: EntropyDiceDuo.ino.teensy40.hex
[upload@6362860-Teensy] Flash usage: 20 kiB (1.0%)
**[upload@6362860-Teensy] Sending reset command**
Received removal notification for device 'HID\VID_16C0&PID_0478\7&113038DD&0&0000'
Received removal notification for device 'USB\VID_16C0&PID_0478\0009B57E'
Remove device 'USB\VID_16C0&PID_0478\0009B57E'
Received arrival notification for device 'USB\VID_16C0&PID_048C\6362860'
Examining device node 'USB\VID_16C0&PID_048C\6362860'
Device 'USB\VID_16C0&PID_048C\6362860' has no 'PortName' registry property
Received arrival notification for device 'USB\VID_16C0&PID_048C&MI_00\7&2B03508F&1&0000'
Examining device node 'USB\VID_16C0&PID_048C&MI_00\7&2B03508F&1&0000'
Found port number of 'USB\VID_16C0&PID_048C\6362860': 3
Found port number of 'USB\VID_045B&PID_0209\5&38BA1A52&0&5': 5
Found controller ID for 'USB\ROOT_HUB30\4&362721B3&0&0': 1
Add serial device 'USB\VID_16C0&PID_048C\6362860' on iface 0
  - USB VID/PID = 16c0:048c, USB location = usb-1-5-3
  - USB manufacturer = Teensyduino, product = Triple Serial, S/N = 6362860
Identified 'Teensy 4.0' with bcdDevice value 0x279
Received arrival notification for device 'USB\VID_16C0&PID_048C&MI_02\7&2B03508F&1&0002'
Examining device node 'USB\VID_16C0&PID_048C&MI_02\7&2B03508F&1&0002'
Found port number of 'USB\VID_16C0&PID_048C\6362860': 3
Found port number of 'USB\VID_045B&PID_0209\5&38BA1A52&0&5': 5
Found controller ID for 'USB\ROOT_HUB30\4&362721B3&0&0': 1
Add serial device 'USB\VID_16C0&PID_048C\6362860' on iface 2
  - USB VID/PID = 16c0:048c, USB location = usb-1-5-3
  - USB manufacturer = Teensyduino, product = Triple Serial, S/N = 6362860
Identified 'Teensy 4.0' with bcdDevice value 0x279
Received arrival notification for device 'USB\VID_16C0&PID_048C&MI_04\7&2B03508F&1&0004'
Examining device node 'USB\VID_16C0&PID_048C&MI_04\7&2B03508F&1&0004'
Found port number of 'USB\VID_16C0&PID_048C\6362860': 3
Found port number of 'USB\VID_045B&PID_0209\5&38BA1A52&0&5': 5
Found controller ID for 'USB\ROOT_HUB30\4&362721B3&0&0': 1
Add serial device 'USB\VID_16C0&PID_048C\6362860' on iface 4
  - USB VID/PID = 16c0:048c, USB location = usb-1-5-3
  - USB manufacturer = Teensyduino, product = Triple Serial, S/N = 6362860
Identified 'Teensy 4.0' with bcdDevice value 0x279

MikeMaurice, 2020-09-02:
I have noticed the red programming light, very dim, not stopping for some time after upload. This sounds a little like your comment above.

Defragster, 2020-09-02:
The 'red LED' time on depends more on the code size and time to erase and flash the upload.
This sketch is only :: Sketch uses 19952 bytes + Global variables use 42076 :: [upload@6362860-Teensy] Flash usage: 20 kiB (1.0%)
{this is on a T_4.0 as it was connected}
That is bigger than the above indicated 14 KiB - and total Red LED active under 1 second.

The referred to RED is a text flash on TyComm UI after upload.  I just uploaded this sketch about 10 times without seeing it. I'll look to make a note next time I see it.

Koromix, 2020-09-03:
I have an idea about your problem but I haven't been able to reproduce it yet.
I'll post a build for you to try shortly.

Koromix, 2020-09-03:
I can't SSH in to my server from where I am right now, so I've temporarily uploaded new Win64 builds here: 

* MSI: https://drop.infini.fr/r/N05sMQMmBa#9aS5LyIf47KEdhHVbsl1KEc1B9JP7l4+ams7yvEnubE=
* ZIP: https://drop.infini.fr/r/472Yf9v7_t#/au9Z0/s023E1QAjC2yW6Oj6djrc6b6t+HxjCFE6PR0=

Defragster, 2020-09-03:
I got a copy.  I'll see if I can repro on the current version and post log text.  Then try that win build.
Not sure what it was about those times/sketches - maybe it was USB input on startup ????

Defragster, 2020-09-03:
Not sure this is it  - but the message may be : "Board '6362860-Teensy' is busy on task 'upload@6362860-Teensy'"

That was generated with a Send after upload on Serial - close after startup.

I need a better repro sample sketch, this using triple serial with a busy Serial  - moved to SerialUSB1 and then I can't trigger it.

I mis-routed the Serial and hung the Teensy and it hung TyCommander to 'Not Reponding' with a quick edit:
	while ( 0 == SerialUSB1.available() );
	while ( SerialUSB1.available() ) {
		SerialUSB1.print( (char)**Serial.read**() ) ;  // should have been  SerialUSB1.read()
	}
I had to task end tycomm.  Still running last release version - and done for today.

closed 2021-01-19 by Koromix