Koromix/tytools #63: Suggestion: Add tytools to platformioto automate use there

blackketter, 2020-10-19:
Not sure if this should be a tytools issue or a platformio one, but thought I'd start here.  It would be great if tytools were a tool available to platformio users directly.

Koromix, 2020-11-20:
That would be cool and maybe it can be done, though I don't know anything about platformio (tried it once and that's it).
I'd like to look into it eventually but available time is very limited right now. So no promises :)

jbliesener, 2021-01-15:
This is how I use tyTools in PlatformIO:

1. Add the following file ```upload_with_tycmd.py``` to the top-level directory of your PlatformIO project (next to ```platformio.ini```):

```
Import("env")

env.Replace(
    UPLOADER="\"c:\\program files\\tyTools\\tycmd.exe\"",
    UPLOADCMD="$UPLOADER upload $UPLOADERFLAGS $SOURCE"
)
```

2. Add this to ```platformio.ini```:

```
extra_scripts = upload_with_tycmd.py
upload_protocol = custom
upload_flags = -B<teensy serial number> 
```

The last parameter (```upload_flags```) is optional and allows to send the sketch to a fixed Teensy, identified by its serial number.

Hope it helps

blackketter, 2021-01-16:
Thanks, @jbliesener, but my purpose for this issue was to suggest that the tytools to be installed by platformio automatically, as they have been more reliable for me than the default tools.

FWIW, I already have tytools working with platformio on macOS with a slightly different technique:
```
[env]
upload_command = tycmd upload -B $UPLOAD_PORT $SOURCE
``` 
where tycmd is already in the $PATH, and for a specific teensy target:
```
[env:<your target name>]
upload_port = <your specific teensy, like "2236890-Teensy">
```