Koromix/tytools #64: Compatibility to upcoming Teensyduino version

luni64, 2020-10-26:
The currently worked on version of Teensyduino changes the so fare used device level CDC class to a composite device. Looks like the TyTools are not happy with this.  The device manager shows the Teensy as e.g. COM3, Teensy.exe uploads correctly,  but the TyTools detect it at COM4 which of course won't work. However, using the Teensy in dual / triple serial mode works normally. 

Here the differences in the generated devices before and after the change. 

```
The old version generates this device USB\VID_16C0&PID_0483\8077220
The new version generates this device: USB\VID_16C0&PID_0483&MI_00\7&922067D&0&0000
```
See here for some more information https://forum.pjrc.com/threads/63695-malloc-free-for-EXTMEM-and-DTCM?p=257329&viewfull=1#post257329

I can send you a .hex showing the effect if you wan't

Koromix, 2020-10-27:
Yes, a .hex would be nice (for T4.0, I don't have a T4.1), or instructions on where to find the version of Teensyduino that does this. I don't really follow Teensy development anymore.

Defragster, 2020-10-27:
This is on Windows 10.  It is based on pending changes on github to CORES.

I built two HEX files that fail on TyCommander - not sure if they demonstrate the problem as you need to see it?

Did a 15s Restore to get them back right ???

Hi Niels.
[smAllocPrint.ino.TEENSY40.zip](https://github.com/Koromix/tytools/files/5443222/smAllocPrint.ino.TEENSY40.zip)

luni64, 2020-10-27:
I'm on Win10. Here, two hex files compiled with the old (current) and the new (upcoming) Teensyduino:

[old_TD.zip](https://github.com/Koromix/tytools/files/5443360/old_TD.zip):
[image]

[new_TD.zip](https://github.com/Koromix/tytools/files/5443382/new_TD.zip)
[image]

Device Manager reports COM8 for new_TD.hex, TyTools detects COM7. 

```c++
#include "Arduino.h"
void setup()
{
  while (!Serial) {  }
  Serial.println("Hello TyTools");
}
void loop(){
}
```

Defragster, 2020-11-01:
@Koromix - do you need anything else to get to the bottom of this?

It seems that change is coming to Teensy and TyCommander fails and having to use the IDE for SerMon is a step backwards.

If you get a version for testing just post

Koromix, 2020-11-20:
Sorry for the delay.

Yeah I could not reproduce it on my T4.0 with your firmware.
I don't know if it has to do with the T4.0 or something else, but I've just bought a T4.1 anyway. It should arrive next wednesday (25/11) and I will resume tests at this time, unless you can reproduce it yourself on a T4.0.

luni64, 2020-11-22:
Sorry for my late answer I was sick the last week. 
I can still reproduce this on a T4. If you load the old version and then the new version Windows will assign a new port number because it sees a new device (composite with one cdc in the new FW opposed to a plain cdc with the old FW) Looks like the tyTools somehow uses the wrong comport. 

Let me know if you want me to test something.

Defragster, 2020-11-22:
Indeed something problematic in the new TD1.54 beta change.  I've been using Dual Serial to work around.

Koromix, 2020-11-23:
> Sorry for my late answer I was sick the last week.
> I can still reproduce this on a T4. If you load the old version and then the new version Windows will assign a new port number because it sees a [...]

OK thanks for the repro recipe, I'll try that tonight (it's 9:30 AM here) and get back to you.

Koromix, 2020-11-24:
The latest commit should fix this (it does on my machine).
You can find a build here: https://koromix.dev/files/tytools/

closed 2020-11-24 by Koromix

luni64, 2020-11-24:
Just tested it, works perfectly for me. 
Thanks a lot

Defragster, 2020-11-24:
Grabbing it to go back to back to normal TyComm usage ...

Defragster, 2020-11-24:
Seems to be working well with single Serial now!

Defragster, 2020-11-25:
@luni64 - grab the last linked exe from here [https://github.com/Koromix/tytools/issues/69](url)

Above test build had a QT fault issue

commit 0fd4738f (2020-11-24): Ignore composite device notifications on Win32

Fixes #64