Koromix/tytools #65: Wildcards or Regular Expressions in Command Line Arguments

peterpolidoro, 2020-10-27:
Thank you for the fantastic tools! I am uploading firmware to 16 Teensy boards connected to one host computer and these are a lifesaver.

I am just curious about how difficult it would be to add support for wildcards or regular expressions in the command line arguments to upload to sets of boards or a board where only part of the location is known. An example might be @/dev/ttyACM[0-1] to upload to both @/dev/ttyACM0 and @/dev/ttyACM1. Or perhaps something like @usb-*-*-1-1 if you wanted to upload to a board connected to a particular USB hub port, regardless of the host computer USB port the hub is plugged into.

Another alternative to modifying the code in this repository is to wrap this code with something like a Python script to parse the command line arguments before calling this code one or more times. There are Python libraries, like sre_yield, that make this easy. I do not know if there is anything similar for c code. If there is, that could be a really cool feature for your tools. If not, then I will just write a little Python wrapper for your code. Thanks!

Koromix, 2020-11-20:
Thanks for your report.

I think it is outside of the scope of tycmd, your wrapper idea is probably a good path.
You can get information about connected teensy devices in JSON format with `tycmd list -O json -v`, and then use the other commands to direct uploads to specific boards, such as `tycmd upload -B @usb-1-2-2 firmware.hex`.
The README contains some information about tycmd features: [https://github.com/Koromix/tytools/blob/master/README.md](https://github.com/Koromix/tytools/blob/master/README.md).

Closing this issue, but feel free to add comments if there is a specific thing missing from tycmd to wrap it correctly.

closed 2020-11-20 by Koromix

peterpolidoro, 2020-11-20:
Thanks I will go ahead and write a Python wrapper for it.

I can compile your code into executables and then call it from Python using your command line interface like you suggested.

It might also be nice, though, if your command line interface was totally separated from the rest of your code. That would provide the other option of compiling a shared library that languages like Python could call directly. You have created a great tool. Having the option of both compiling executables and shared libraries would make it easily reusable from lots of other languages. I do not know if that would be lots of work for you or relatively easy, but it is just a suggestion. Thanks again for all of your nice work on this!