Koromix/tytools #66: Firmware is not compatible

avpanasyuk, 2020-11-04:
Hello, Koromix

Thank you very much for your tools!

 I have an issue when I try to program  TEENSY++ 2.0, though. The board is properly recognized as "TEENSY++ 2.0" in bootloader mode, but when I try to upload firmware that "Firmware ... is not compatible with '12345-Teensy'. It does not work for either ELF or standalone HEX files. 

The firmware successfully loads with Teensy Loader and works file. 

Log file contains the same message, but no additional information.

I would appreciate any insights. An option to disable this compatibility check would be very helpful, as it seems to be not a new issue.

Best regards
Alex.

Koromix, 2020-11-20:
Thanks for the report,

Could you send me the HEX file if you don't mind?
You can do it here or on my mail: niels.martignene@protonmail.com

Regards,
Niels

Koromix, 2021-01-19:
Closing this issue due to lack of information.

closed 2021-01-19 by Koromix