Koromix/tytools #67: TyCommander Log File status

Defragster, 2020-11-22:
On Windows 10 with T_4.1:
I've been looking to the disk copy of the Serial data and seeing them not matching to the current state of the GUI display to the 'newest' tail of the data?

And a Teensy can go offline and the saved file is not updated to show the last received data?

Not sure if that is a setting on my end or a missing flush to file or something I can do to force that when the Teensy is no longer pushing data is when I want to see the larger file for history that paged off - and it fails to show the current tail of the GUI data.

It seems this used to work - maybe this is a side effect of  speeding up the GUI to keep up with faster 480 Mbit/sec USB?

Koromix, 2020-11-23:
The log file is capped to a certain size. When this size is reached, it starts overwriting the beginning.

TyCommander write a delimiter line (@@@@@@@@@@@@@@@@@@@@@@@@@@@ [...]) to help you locate the boundary between new and old content.

Defragster, 2020-11-23:
I see multiple files for each Teensy #1,2,3,4 :: TyCommander-7658860-Teensy-4.txt

I thought it would close a file and continue into the next.  - that would be awkward as well of course.

I did not know the '@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@' trick!
Yes - there is the end of the file.

closed 2020-11-23 by Defragster

Koromix, 2020-11-23:
If I remember correctly, a new file is used when the Teensy appears or TyCommander is launched. While a board is being used, it stays on the same file and uses the @@@@@ delimiter thingy. I think :)

Maybe it is not the smartest thing to do and rotating files would be better. I'm open to it; any though on this?

Defragster, 2020-11-23:
Rotating files WOULD start file empty and then show (for example) three lines of new output - with nothing before it until the prior file was found.

That isn't clean either.  Knowing to search for "@@@@@@" is easy, except for being a secret :)

I'll keep an eye on it and make a note if that doesn't seem right.

Hope you are staying healthy and enjoying your job.

Koromix, 2020-11-23:
That was the idea, I think.

Now if filesystem APIs were better, it would be easy to "truncate the beginning" of a file and so we could cap the log size AND keep the latest data at the end. But we don't have that unfortunately.