Koromix/tytools #68: Restart timing on upload?

Defragster, 2020-11-22:
Is there a built in expectation constraint on upload reset?

New LittleFS can use program FLASH for a file system - which forces it to do a full format before the upload completes and this takes a long time - even with a small sketch.

Have seen - perhaps noted in anothe rissue - some time the Teensy is seen as not available with some error RED noted - and TyCommander gets out of sync with the Teensy and a button press gets it back to recognized and a repeat upload can work.

I've actually gone back to Teensy Loader and the new --delegate flag allowing command line builds and Teensy Loader to perform the uploads so not investigated much.

Koromix, 2020-11-23:
Could you send me a non-compiled sketch that will trigger this, along with information on which Arduino+Teensyduino version to test it with?

I don't follow Teensy development anymore, and unfortunately I don't have much time to dig around.

Defragster, 2020-11-23:
Yes, I will try to get a sketch that will repro.  The worst case was TD 1.54 beta using the LittleFS as a file system using space on the PCB Mounted FLASH chip.  You can try it when you get your T_4.1 in coming days - I'll set up a sketch and see if it acts like it did before with the latest Beta.

Luckily TyCommander is in Good shape - miss seeing you around the forum.

Defragster, 2020-12-19:
Closing this as it must have been a conflict with earlier Beta code before last TyCommander update?

I have done some of the uploads that it seemed caused this and they worked fine.  SO not seen this as a problem in many recent uploads.

closed 2020-12-19 by Defragster