Koromix/tytools #69: Win10 multiple Teensy : one not programmed Hover on Recent faults

Defragster, 2020-11-24:
Newest ver - 3 Teensy only one uploaded so far.

Go to one not programmed and right click and HOVER over the 'Empty' "Upload recent firmware'

In seconds the TyCommander app vanishes.

Restart Tycommander with no upload - and it repeats with right click and hover.

Last time I tried this it seemed any firmware uploaded (to any Teensy) would be shown?

Koromix, 2020-11-24:
Yes I've seen the same thing when trying the dev build.
I'll fix it for the next one.

Defragster, 2020-11-24:
:)

Koromix, 2020-11-24:
It's a Qt bug: https://bugreports.qt.io/browse/QTBUG-85758
I'll make a new build TyTools with a Qt version that works correctly and post it soon.

Defragster, 2020-11-24:
Sounds good, Thanks - no rush - I'll just stop doing that for now.

Koromix, 2020-11-24:
I've pushed a new build, with Qt 5.12.10, that seems to fix the issue.
https://koromix.dev/files/tytools/

closed 2020-11-24 by Koromix

Defragster, 2020-11-24:
Looks good here!