Koromix/tytools #70: TyUpdater 0.9.3 can't be opened on macOS

2xAA, 2021-01-12:
Tested on macOS 10.14 and macOS 11 - both have the same error.

TyUpdater 0.9.2 works perfectly.

[image]

Koromix, 2021-01-16:
Weird.. I'll try it today, though I only have El Capitan here (Macbook is from 2009), and look for what differs between 0.9.2 and 0.9.3.

Koromix, 2021-01-19:
Yeah, I should have noticed the problem when I made the release... The dmg file is ten times smaller than the one of 0.9.2. That's because I made the mistake of building with a non-static build of Qt.

I'm rebuilding Qt and then TyTools, it should be ok in a few hours (MacBook from 2009, HDD... it's slow).

Koromix, 2021-01-19:
I've replaced the DMG with a working one, thanks!

closed 2021-01-19 by Koromix