Koromix/tytools #73: Issue during upload

CWanjohi, 2021-02-05:
I am using Teensy 4.0 on Kali Linux with tycmd running on terminal.
This is the command I am executing and the output on terminal as follows:
```tycmd upload Teensy_U2F.hex --board 8709100@usb-1-2```
       upload@8709100-Teensy  Uploading to board '8709100-Teensy' (Teensy 4.0)
       upload@8709100-Teensy  Firmware: Teensy_U2F.hex
       upload@8709100-Teensy  Flash usage: 505 kiB (25.4%)
       upload@8709100-Teensy  Uploading... 100%
       upload@8709100-Teensy  Sending reset command (with RTC)
       upload@8709100-Teensy  Failed to reset board '8709100-Teensy'
       
Kindly assist, thanks.

Defragster, 2021-02-05:
to repro I suspect the HEX file would be needed.

Do other HEX files upload and work properly?

Does it fail the same when RTC not set ... would require prior Release version as the option to disable not complete yet.

Koromix, 2021-05-05:
No news since Defragster's questions, so I'm closing this issue.
Feel free to open a new one in case this issue is still relevant.

closed 2021-05-05 by Koromix