Koromix/tytools #78: Couldn't upload firmware

lavanthaya, 2021-06-23:
Hi,

I am trying to use tytools from the Ubuntu machine to upload firmware to the teensy board (3.2),
But it's not happening.
With tytools, I can list connected devices, can monitor but upload and reset is not working,
I'm getting below error,

[image]

Can anyone please help me with this?
Note:I am running an ubuntu machine on a VMware workstation (VM).

Koromix, 2021-06-24:
Which version ot TyTools are you using?
Type `tycmd --version` to check.

lavanthaya, 2021-06-24:
My version is,
tycmd 0.9.3-7-g766531b

Koromix, 2021-06-24:
Did you follow the Linux instructions to setup udev correctly?
You can find them here: https://www.pjrc.com/teensy/td_download.html (__Linux installation__ part)

lavanthaya, 2021-06-24:
Yes, I followed the same instruction to install teensy,
As I mentioned above, I can use tycmd command with "list" or "monitor", only the "upload" or "reset" commands are not working

Koromix, 2021-11-02:
I just noticed the use of VMWare.

When a Teensy is programmed, the device disappears and then reappears. If you don't automatically (or quickly) ressasign it to the VM, it looks like it is gone from inside the VM.

I think that explains the problem you are seeing. I'm closing this issue for this reason, feel free to reopen if I'm mistaken.

closed 2021-11-02 by Koromix