Koromix/tytools #82: Teensy 4.x w/1062 coming out in Encrypted version

Defragster, 2021-11-03:
Currently in Beta - only Teensy.exe Loader knows the process and steps for parsing an encrypted .eHex file and determining when it is required for upload and then programming it into a 1062 Teensy.

First step would be having TyComm 'delegate' uploading as a .hex file will not work when locked and an .eHex is required.

Currently using TyComm for SerMon and doing 'Ctrl+R' verify builds then with Teensy.exe active I hit 'gui Bootloader' to drop the USB connection and put Teensy in Bootloader which inspires the Teensy>exe to upload.

NOTE: the "Reset" process of triggering bootloader than triggering start really 'confuses' Teensy.exe as it catches the bootloader event - then gets blocked by TyComm and Teensy.exe does odd timeout when the Teensy disappears and starts running.

Also there isn't a single thread with all the Locked 1062 info - it started with TD 1.55 and is extending into TD 1.56 b1 and b2 so far.  Expected hardware availability before December 2021.

Koromix, 2021-11-03:
Thanks for the info, will eventually look into it :)

Defragster, 2022-01-07:
Encrypted 1062's are soon if not already online - DIY bootloader chips have chipped - even it seems SFun MicroMod making one too?
Not sure if updated Teensy Loader to expose the steps is pending or do-able. When locked it seems not only must the HEX key be identified as needed - but perhaps a stub code needed to interface to the locked chip in some fashion?
To date have only tried using Teensy.exe for uploads - not sure what error TyComm might see.
Have been using IDE without integration or TSET from editor windows batch file with --delegate. Not tried 'integration with --delegate' option?
Perhaps a Teensy 1062 option by Serial# to 'delegate' uploads? Can the TeensyCLI tool be called so that loader isn't needed?

tluliak, 2022-01-16:
Paul has posted some additional information on the .ehex file format [here](https://www.pjrc.com/teensy/td_code_security.html#:~:text=Teensy%204.0%2C%20Teensy%204.1%2C%20and,in%20the%20flash%20memory%20chip.). Is there currently a version of TeensyCLI that will write .ehex to locked boards?

strud, 2025-02-28:
I'm also keen to understand if TyCommander will likely implement ehex support.

EvanCox14, 2025-12-16:
In case this helps anybody else:
I was able to upload an encrypted verifyIno ehex file using `C:\{path}\tycmd.exe upload C:\{path}\KeyVerify.ino.ehex -f ihex`, but that was before I had locked it using the lockIno code. It passed the encryption checks when it ran. I confirmed that it wasn't a false positive by uploading the verifyIno hex file which failed the encryption checks.
Now that I've securely locked it with the lockIno ehex, I have been unable to reproduce this success with tycmd and confirmed that I am still able to upload ehex files with arduino IDE. If anyone knows of how to get tycmd to upload ehex files onto a secure locked teensy, I am all ears.

Koromix, 2025-12-17:
Until I can make a more complete release with the GUIs (which requires some Qt upgrades... and a an Apple machine for the macOS version which I don't have right now), you can try tycmd with the Windows x64 build below:

[tycmd_0.9.10-beta.1_windows_x64.zip](https://github.com/user-attachments/files/24215012/tycmd_0.9.10-beta.1_windows_x64.zip)

Koromix, 2025-12-19:
I've added a longer delay between HAB writes, and added retry and delay logic for transient I/O errors when sending the booloader to the HAB. I've made a new beta release (below), which should help with the I/O errors (hopefully).

The other errors are less surprising and could come from the I/O errors. For example, when the bootloader send fails because of an I/O error, the Teensy is left in "HAB mode". When tycmd runs again, all it finds over the USB is the HAB... which does not expose the serial number. Which is why it thens reports seeing a "?-Teensy" board. Because it does not have access to the S/N in this mode.

[tycmd_0.9.10-beta.2_windows_x64.zip](https://github.com/user-attachments/files/24254262/tycmd_0.9.10-beta.2_windows_x64.zip)