Koromix/tytools #83: Teensy support RawHid 512 byte, not 64 byte

Defragster, 2021-11-08:
The T_4.x can do RawHid 512 byte interface - 8X faster than the 64 byte.

KurtE has a WIP that works. In that code a Serial.print goes into PRJC Teensy Serial Monitor code, but it does not appear in TyCommander?

TyComm sees the device - but does not show any output from it.
Here is what the 'log' shows on powering it up in this mode:
`Received arrival notification for device 'USB\VID_16C0&PID_0486\9706430'
Examining device node 'USB\VID_16C0&PID_0486\9706430'
Ignoring composite device
Received arrival notification for device 'HID\VID_16C0&PID_0486&MI_00\8&29F970D8&0&0000'
Examining device node 'HID\VID_16C0&PID_0486&MI_00\8&29F970D8&0&0000'
Found port number of 'USB\VID_16C0&PID_0486\9706430': 2
Found port number of 'USB\VID_0BDA&PID_5411\5&12668A1&0&6': 6
Found controller ID for 'USB\ROOT_HUB30\4&3765104F&0&0': 2
Add HID device 'USB\VID_16C0&PID_0486\9706430' on iface 0
  - USB VID/PID = 16c0:0486, USB location = usb-2-6-2
  - USB manufacturer = Teensyduino, product = Teensyduino RawHID, S/N = 9706430
  - HID usage page = 0xffab, HID usage = 0x200
Identified 'Teensy 4.1' with bcdDevice value 0x280
Received arrival notification for device 'HID\VID_16C0&PID_0486&MI_01\8&1235E493&0&0000'
Examining device node 'HID\VID_16C0&PID_0486&MI_01\8&1235E493&0&0000'
Found port number of 'USB\VID_16C0&PID_0486\9706430': 2
Found port number of 'USB\VID_0BDA&PID_5411\5&12668A1&0&6': 6
Found controller ID for 'USB\ROOT_HUB30\4&3765104F&0&0': 2
Add HID device 'USB\VID_16C0&PID_0486\9706430' on iface 1
  - USB VID/PID = 16c0:0486, USB location = usb-2-6-2
  - USB manufacturer = Teensyduino, product = Teensyduino RawHID, S/N = 9706430
  - HID usage page = 0xffc9, HID usage = 0x4
Identified 'Teensy 4.1' with bcdDevice value 0x280`

This is what a T_4.1 shows in the log after programming as a 64 byte RawHID device, that works to show the Teensy Serial.print():
`[upload@9706370-Teensy] Waiting for Teensy Loader
Received removal notification for device 'HID\VID_16C0&PID_0478\8&24893BEA&0&0000'
Received removal notification for device 'USB\VID_16C0&PID_0478\000ECF8D'
Remove device 'USB\VID_16C0&PID_0478\000ECF8D'
Received arrival notification for device 'USB\VID_16C0&PID_0486\9706370'
Examining device node 'USB\VID_16C0&PID_0486\9706370'
Ignoring composite device
Received arrival notification for device 'HID\VID_16C0&PID_0486&MI_00\9&C8C615A&0&0000'
Examining device node 'HID\VID_16C0&PID_0486&MI_00\9&C8C615A&0&0000'
Found port number of 'USB\VID_16C0&PID_0486\9706370': 1
Found port number of 'USB\VID_0BDA&PID_5411\6&1701749F&0&1': 1
Found port number of 'USB\VID_0BDA&PID_5411\5&12668A1&0&6': 6
Found controller ID for 'USB\ROOT_HUB30\4&3765104F&0&0': 2
Add HID device 'USB\VID_16C0&PID_0486\9706370' on iface 0
  - USB VID/PID = 16c0:0486, USB location = usb-2-6-1-1
  - USB manufacturer = Teensyduino, product = Teensyduino RawHID, S/N = 9706370
  - HID usage page = 0xffab, HID usage = 0x200
Identified 'Teensy 4.1' with bcdDevice value 0x280
Received arrival notification for device 'HID\VID_16C0&PID_0486&MI_01\9&244FED9F&0&0000'
Examining device node 'HID\VID_16C0&PID_0486&MI_01\9&244FED9F&0&0000'
Found port number of 'USB\VID_16C0&PID_0486\9706370': 1
Found port number of 'USB\VID_0BDA&PID_5411\6&1701749F&0&1': 1
Found port number of 'USB\VID_0BDA&PID_5411\5&12668A1&0&6': 6
Found controller ID for 'USB\ROOT_HUB30\4&3765104F&0&0': 2
Add HID device 'USB\VID_16C0&PID_0486\9706370' on iface 1
  - USB VID/PID = 16c0:0486, USB location = usb-2-6-1-1
  - USB manufacturer = Teensyduino, product = Teensyduino RawHID, S/N = 9706370
  - HID usage page = 0xffc9, HID usage = 0x4
Identified 'Teensy 4.1' with bcdDevice value 0x280`

This doesn't look to add anything - but is what PJRC teensy_ports.exe shows:
`C:\T_Drive\Arduino_1.8.15td155\hardware\tools>teensy_ports.exe
{
  "address": "usb:0/140000/0/6/2",
  "online": true,
  "label": "hid#vid_16c0&pid_0486 (Teensy 4.1) RawHID",
  "vid": "16C0",
  "pid": "0486",
  "iserial": "970643",
  "boardName": "Teensy 4.1",
  "protocol": "Teensy"
}`

@KurtE , @mjs513 : not sure if there is anything to add about the difference to support that may be needed? The PORT or PAGE number or ???

Koromix, 2021-11-08:
I'll look over this next week-end :)
How do I enable this new Raw HID mode?

KurtE, 2021-11-08:
Warning: So far this is only an experiment.  As Paul has sanctioned it, or commented on it, and not sure if he has even read the thread talking about it...  But 8 times the throughput feels like a good thing. 

Some details up on the thread: https://forum.pjrc.com/threads/68620-T4-x-support-for-Raw-HID-512-wonder-if-it-makes-sense-to-add-to-system

Note: as I have it, it still uses the same VID/PID of the RAWHID USB type.  Which sets up two interfaces:
RAWHID and SEREMU (Serial emulator)

Again this only can work on T4.x not T3.x 
There are two differences in the setup for 512... 
If it connects at High speed connections (480mbs) packet sizes are 512. So in this definition for the end point is setup that for the RAWHID interface the Max packet size is set to 512.  If you connect at slower setup (12mbs), packet sizes are 64 bit like T3.x and the Max packet sizes are set to 64... 

My version of Pauls HID PC code (in my rawhid github project) On linux side so far I detect the differences looking at end point. 

On the windows version, I hacked it up that I changed the HID report to show the report size is 512, which could be wrong if we connect at a low speed... 

So as for tyCommander, not sure what the impact would be?  It should still see the same SEREMU interface and I would doubt that you would do anything with the RAWHID stuff?

Defragster, 2021-11-08:
Is it a matter of ignoring RawHid when normal Serial is presented?   Or maybe adding a second interface ( like Dual Serial ) for both RawHid and Serial?

Rebuild of sketch above from 64B RawHid to 'Serial' shows this:
`[upload@9706370-Teensy] Waiting for Teensy Loader
Received removal notification for device 'HID\VID_16C0&PID_0478\8&24893BEA&0&0000'
Received removal notification for device 'USB\VID_16C0&PID_0478\000ECF8D'
Remove device 'USB\VID_16C0&PID_0478\000ECF8D'
Received arrival notification for device 'USB\VID_16C0&PID_0483\9706370'
Examining device node 'USB\VID_16C0&PID_0483\9706370'
Ignoring composite device
Received arrival notification for device 'USB\VID_16C0&PID_0483&MI_00\8&6E867D8&0&0000'
Examining device node 'USB\VID_16C0&PID_0483&MI_00\8&6E867D8&0&0000'
Found port number of 'USB\VID_16C0&PID_0483\9706370': 1
Found port number of 'USB\VID_0BDA&PID_5411\6&1701749F&0&1': 1
Found port number of 'USB\VID_0BDA&PID_5411\5&12668A1&0&6': 6
Found controller ID for 'USB\ROOT_HUB30\4&3765104F&0&0': 2
Add serial device 'USB\VID_16C0&PID_0483\9706370' on iface 0
  - USB VID/PID = 16c0:0483, USB location = usb-2-6-1-1
  - USB manufacturer = Teensyduino, product = USB Serial, S/N = 9706370
Identified 'Teensy 4.1' with bcdDevice value 0x280`

Koromix, 2021-11-14:
I've just pushed some code to read the expected RAWHID report size from the HID descriptors (either preparsed by the OS on Windows and macOS, or retrieved from the HID descriptor on Linux).

No idea if any of this works, I'm not using my usual computer and I don't even have a compiler on this one. I'll how this goes when I get home :)

Defragster, 2021-11-14:
Cool, @KurtE probably distracted on other issues.

Kurt said it does Seremu and RawHid.  If so then when it presents as RawHid512 having that is one thing - but should it also open a 2nd link to the Seremu? where it seems the Serial.print() output then goes?

KurtE, 2021-11-14:
Sorry been busy with other things…

 

Not sure at this point if/why the Tytools looks at Rawhid?  Seremu is sort of a rawhid, but has specific thing.  I would think this would act the same way as when configured for MTP or Joystick… 

And probably that interface is simply ignored.

Defragster, 2021-11-14:
When Teensy is ONLY RawHID - like Teensy_Serial - it needs to pick that up to see anything.

But when Seremu is possible it seems in this case it is deferring to RawHid and ignoring that?

Koromix, 2021-11-14:
> Sorry been busy with other things… Not sure at this point if/why the Tytools looks at Rawhid? Seremu is sort of a rawhid, but has specific thing. I would think this would act the same way as when configured for MTP or Joystick… And probably that interface is simply ignored.

Well, yes. For some reason I was this issue and went full ahread with dynamic SEREMU report size instead, which is what Ive done in the branch on the train :)

Still, this could maybe solve the issue a bit by accident. I'll look into the actual problem in a few days with a fresh mind ;)

Koromix, 2021-11-20:
I've tested a bit and I can't reproduce any problem with these dev builds:
https://koromix.dev/files/tytools/

closed 2022-12-14 by Koromix