Koromix/tytools #84: Large HEX T_4.1 won't pass TyCommander for upload: is too big to load

Defragster, 2021-12-30:
This sketch: https://forum.pjrc.com/threads/69113-Initialize-Variable-in-SPI-Flash?p=296621&viewfull=1#post296621

Seems to get checked by TyCommander and upload refused:
teensy_size: Memory Usage on Teensy 4.1:
teensy_size:   FLASH: code:26824, data:4199320, headers:9116   free for files:3891204
teensy_size:    RAM1: variables:5632, code:24152, padding:8616   free for local variables:485888
teensy_size:    RAM2: variables:12384  free for malloc/new:511904
Waiting for user selection
Firmware 'R:\temp\\arduino_build_AllocProgmem.ino\AllocProgmem.ino.hex' is too big to load

A button push allows Teensy Loader to program, or working from IDE without integration

Koromix, 2021-12-31:
Thanks for the report.

This should work better with these builds: https://koromix.dev/files/tytools/
I have only done light testing with them :)

Regards,
Niels

Defragster, 2021-12-31:
Thanks Niels, downloading now - will give it a try.

@KurtE @mjs513

Koromix, 2022-01-01:
Did some more testing, I could not find any problem.
I will make a small release soon unless someone encounters a bug.

Defragster, 2022-01-01:
Just tested and this worked! Bumped up to using 7MB of data!
Only issue is TyCommander is too quick to call 'Failed to reset'
 > large prior code takes a long time to format Flash
 > large current code takes even longer to upload
 > Also then TyComm doesn't save the name of the last HEX upload?

Note: Teensy 1062's got a new bootloader that no longer wipes ALL flash before programming - only min 512KB (1024 is a LOCKED) or if larger the size of the HEX in 64KB steps. 1062 are T_4 is 2MB, T_4.1 is 8MB and T_MM is 16MB

Maybe this is a new issue if it is something that can be resolved? But I think not. The LOCKED Teensy needs upload using Teensy loader - so not Integrating TyComm but using the --delegate option from my Batch file from the editor. 
 > GUI RESET no longer works nicely as active Loader tries to Program
 > TyComm steals the Teensy from the loader when it thinks it is programming it.

 > This large data sketch worked:
teensy_size: Memory Usage on Teensy 4.1:
teensy_size:   FLASH: code:26952, data:7346072, headers:8988   free for files:744452
teensy_size:    RAM1: variables:6656, code:24280, padding:8488   free for local variables:484864
teensy_size:    RAM2: variables:12384  free for malloc/new:511904
Waiting for user selection
       upload@9706370-Teensy  Uploading to board '9706370-Teensy' (Teensy 4.1)
       upload@9706370-Teensy  Triggering board reboot
       upload@9706370-Teensy  Waiting for Teensy Loader
       upload@9706370-Teensy  Failed to reset board '9706370-Teensy'

closed 2022-01-01 by Defragster