Koromix/tytools #85: TyCommander doesn't wait long enough for bootloader on 1062's

Defragster, 2022-01-07:
The 1062 Teensy's take longer to program in many cases resulting in

> upload@9706370-Teensy Uploading to board '9706370-Teensy' (Teensy 4.1)
upload@9706370-Teensy Triggering board reboot
upload@9706370-Teensy Waiting for Teensy Loader
upload@9706370-Teensy Failed to reset board '9706370-Teensy'

This can be due to long Flash erase time, or larger code, or when Teensy with MTP USB support takes a long time to come back online.

Result is that message and TyComm not associating the name of the INO with that Teensy as it was not seen as uploaded.

Often it seems TyComm notes the Teensy still in 'bootloader', though that may not indicate that Teensy.exe is still at work - it does when Erase and upload take longer.

Koromix, 2022-01-14:
I've doubled the timeout in the latest commit.
You can find a build here: https://koromix.dev/files/tytools/

Do this help?

Defragster, 2022-01-14:
I'll give it a try.
It isn't all the time - but some sketches just take longer.
IIRC - Paul noted adding a longer wait.

Koromix, 2022-01-17:
Assuming it's fixed with the increase, feel free to commit if not.

closed 2022-01-17 by Koromix

Defragster, 2022-01-17:
Thanks, Yes, thought I posted back.

The longer delay has not failed with the sketches I've been doing!