Koromix/tytools #86: TyCommander - Teensy SEREMU - satisfy while(!Serial) send magic message....

KurtE, 2022-01-17:
@Koromix  @defragster - 

Would be great if TyCommander would send the Magic message to SEREMU when it is logically connected up to a Teensy that is using SEREMU as the serial communications.  

I don't know how to explain the  best place for it, but it should be sent as soon as you are connected up to that Interface and you are ready to do communications with the endpoints. 

The message that needs to be sent is almost identical to the one you send to do the reboot: (seremu_magic
except that the last two bytes are different...

That is if you look at the teensy code:
```#ifdef SEREMU_INTERFACE
	if (setup.word1 == 0x03000921 && setup.word2 == ((4<<16)|SEREMU_INTERFACE)) {
		if (endpoint0_buffer[0] == 0xA9 && endpoint0_buffer[1] == 0x45
		  && endpoint0_buffer[2] == 0xC2 && endpoint0_buffer[3] == 0x6B) {
			printf("seremu reboot request\n");
			usb_start_sof_interrupts(NUM_INTERFACE);
			usb_reboot_timer = 80; // TODO: 10 if only 12 Mbit/sec
		} else {
			// any other feature report means Arduino Serial Monitor is open
			printf("seremu online\n");
			usb_seremu_online = 1;
		}
	}
#endif
```
Anything but the c2 6b...

Thanks
Kurt

Koromix, 2022-01-17:
I've pushed a new build that should do just this. I haven't tested it :)
Here: https://koromix.dev/files/tytools/

KurtE, 2022-01-17:
Appears to work great!

Thanks

Defragster, 2022-01-17:
New TyCommander works.

@KurtE - not seeing a change in startup time?  Does that help with something else? Or need a follow on change in CORES?

Defragster, 2022-01-17:
... just getting to forum to read more ...

closed 2022-01-17 by Koromix

KurtE, 2022-01-17:
Yes, suppose I have up mtp sketch and do a reset…  The one I had some serial outputs:

+++ before MTP.begin() 352

 

 

*** Start Interval Timer ***

+++ before wait on !Serial 352

ERROR: intervaltimer received command with 65535 bytes

ERROR: intervaltimer received command with 65535 bytes

ERROR: intervaltimer received command with 65535 bytes

ERROR: intervaltimer received command with 65535 bytes

timer:652 CMD: 1002(OPEN_SESSION)l: 16 T:0 : 1

652 RESP:2001(RSP:OK)l: 12 T:0

timer:702 CMD: 1001(GET_DEVICE_INFO)l: 12 T:1

GetDeviceInfo size=211

702 RESP:2001(RSP:OK)l: 12 T:1

timer:752 CMD: 1014(GET_DEVICE_PROP_DESC)l: 16 T:2 : d402

752 RESP:2001(RSP:OK)l: 12 T:2

+++ after wait on !Serial 900

 

That last one 900 was after the while (!Serial && millis() < 5000) ;

And before it would always be > 5000