Koromix/tytools #87: Doesn't work for uploading on Teensy MicroMod

collin80, 2022-01-18:
I have a special project that includes a Teensy MicroMod. I have the MM in Dual Serial mode. There are two basic problems here it looks like:

1. It detects the second serial port first. I'm not sure if this matters or not but it seems to try to use the second port preferentially whereas it seems the first port would be more appropriate? At least for me, I'd expect and want to connect to the first port. 
2. All your programs seem to fail with: "I/O error while writing to '/dev/hidraw4': Broken pipe" any time I try to do an upload. The Teensy goes into bootloader / Halfkay mode by this point but no matter how many times I try it always says it had an I/O error on hidraw4 and that there is a broken pipe. Both the normal Teensy GUI upload window and the Teensy CLI tool can flash the board.

Defragster, 2022-01-18:
Something is not right?

Compiled for T_MM Dual Serial and it added second device icon with RED button.
Then recompiled same sketch for MTP and, even if power cycled, both device ports are shown and the SerMon data - from Seremu used with MTP - can appear on either device?

This same behavior now appears on T_4.1. Dual Serial added a second device, and it survives going to MTP w/Seremu and output - only sent to 'Serial' in code may appear on either instance.

There is something odd - may be the sketch I've been using on T_4.1 doesn't like T_MM ...

I closed TyCommander and reopening with MTP w/SEREMU T_MM and T_4.1 work properly, but once it sees Dual Serial changing to MTP w/SEREMU doesn't delete the second device.

I can't say I've done this before to rule out recent code change?

@collin80 - can you try a prior release of TYQT?

videojedi, 2022-01-18:
can confirm similar issue when uploading to a 4.1

[upload@11058150-Teensy] I/O error while writing to '\\.\HID#VID_16C0&PID_0478#6&31027823&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}'

Koromix, 2022-01-23:
Thanks for the report.

This is a regression I introduced in the latest release (0.9.6).
It is fixed in release 0.9.7: https://github.com/Koromix/tytools/releases/tag/v0.9.7

If not, feel free to comment and I will reopen the bug ;)

closed 2022-01-23 by Koromix