Koromix/tytools #88: Problem Uploading Teensy 4.1 with tycmd  >> build_flags = -D USB_MIDI_SERIAL

8JupiterMoll8, 2022-01-23:
Im trying to upload via tycmd in platformio my .hex to teensy 4.1. This works ok. 
But the build_flags = -D USB_MIDI_SERIAL is not set and i got a failure. 
So that teeny can't behave as Midi - Device.
Any ideas to solve this problem?
Thx.

Windows 10
TyTools 0.9.6
PlatformIO : Core 5.2.4  Home 3.4.0

platformio.ini:
[env: LEFT TEENSY]
platform = teensy
board = teensy41
build_flags = -D USB_MIDI_SERIAL
framework = arduino
upload_command = tycmd upload -B $UPLOAD_PORT $SOURCE
upload_port = 10271220-Teensy

Terminal:
      upload@10271220-Teensy  Uploading to board '10271220-Teensy' (Teensy 4.1)
      upload@10271220-Teensy  Triggering board reboot
      upload@10271220-Teensy  I/O error while writing to '\\.\HID#VID_16C0&PID_0478#7&33246076&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}'
*** [upload] Error 1
========================

Koromix, 2022-01-23:
Thanks for the report.

This is a regression I introduced in the latest release (0.9.6).
It is fixed in release 0.9.7: https://github.com/Koromix/tytools/releases/tag/v0.9.7

If not, feel free to comment and I will reopen the bug ;)

closed 2022-01-23 by Koromix

videojedi, 2022-01-31:
Hello, I can confirm TyCommander works... upload,serial, reset etc...

but TyUploader (on windows at least) seems to upload correctly, but the program doesnt start running, even if I push reset or power cycle the Teensy.

However if i start Tycommander or Arduino IDE the teensy immediatley starts running the prog again...

Koromix, 2022-02-03:
By any chance, did you put something like `while (!Serial);` in the setup function, or something similar that could cause the Teensy to wait until a serial monitor (such as the one in TyCommander) is open?

videojedi, 2022-02-03:
Yes that was exactly it! Many thanks........ i feel really silly now....

Great app by the way...

Richard