Koromix/tytools #93: No method to uninstall

bdlabitt, 2022-06-05:
I made the mistake of doing sudo make install of tytools.  There doesn't seem to be a way to uninstall tytools.  Requesting a way to do this.

There seems to be a very strong correlation with tycommander and hard crashing of my laptop.  It occurs after flashing a Teensy.  There are no breadcrumbs in syslog, nor dmesg.  Just a USB access and a hard fault causing my laptop to turn power off.  I recompiled with debug settings.  I did not see a way to make sure optimization was turned off, ie. -O0.  Running with valgrind showed there were memory leaks running tycommander, however, I do not know if some of the errors reported are false due to the optimization level.  How can one tell cmake to set optimization to O0?  Some of the leaks were related to uninitialized variables.

tycommander has been crashing my laptop since 26 May 2022.  Prior to that date, it was working ok.  On that date there was an Ubuntu 20.04 LTS apt update.  Then the crashes happened when ever I tried to program the Teensy.  Since that time, I have done a fresh install on my machine to POP-OS 22.04.  I am still getting Teensy flash related crashes using ty commander.  

I have uninstalled Arduino, and deleted the directory.  Then reinstalled Arduino, along with Teensyduino.  I don't know what to do at this point.  I'd like to totally eliminate tytools from my system, at least for the moment.  It is unfortunate, because when it was working it was very nice.  But I have to stop the violent crashes, before my laptop file system is damaged.  My syslog has been corrupted due to one of the crashes.  Honestly, I really don't know what to do.  This is on an i7 system running 64 bit OS with 32GB RAM and 1TB SSD, with a fresh OS install.  Would appreciate any help on this.  Thanks.

bdlabitt, 2022-06-07:
Today, I got the basic Arduino/Teensyduino working properly for 5 burns of a Teensy 4.1.  I then used tycommander and attempted to "enable integration with Arduino" for a test.  As soon as I clicked the menu item "enable integration with Arduino", my computer froze for 20 seconds, hard crashed and turned off the laptop power.  I have no idea why this app is doing this.  Violent crash with strings of 00's written to syslog, pretty bad if you ask me.  

Weirdly, tycommander works fine on RaspiOS-64 bits.  Find it hard to believe that both Ubuntu 20.04 and Pop!OS 22.04 (fresh install) crash like this, but a RPI4-8GB doesn't crash using tycommander.  Would be nice to get a response of some sort, like, please provide x, y, & z, so maybe we could help you.  I'm not a software developer, so I am not familiar with what is needed to do this kind of detective work.  I like the tool, and find it useful.  Crashing, on the other hand, is not good.

Defragster, 2022-06-07:
@koromix - if it helps for more info posts by clinker8 on forum are [https://forum.pjrc.com/threads/70416-Looking-for-some-clues-hard-Linux-crash-using-Ty-Commander?p=307459#post307459](url)

bdlabitt, 2022-08-04:
For a while, I was able to use tycommander by reverting the kernel to 5.15.0-37.  This did work.  Today was was forced into a dist-upgrade and lost access to this kernel.  Now at kernel 5.18.10-76051810-generic and my laptop turns off 1 minute and 8 seconds after programming a Teensy4.1.  

So, to get on with life, I think I need to break the integration of Arduino with TyCommander.  I just can't use the TyCommander on a fully updated PopOS/Ubuntu based system because it causes full kernel panic with zero traces.  No one can seem to help me, so I have to dump this tool.

How do I sever the integration of this tool with Arduino?  I tried resetting the app, but that did not work.
I tried launching the app by itself while the USB port was active (but not Arduino) and my computer froze for 1 minute and turned off.  After I turned on the computer again, I started Tycommander and reset the app and settings.  This did not break the connection.  When attempting to burn a Teensy, tycommander came up.  I want the Teensyloader again.

Defragster, 2022-08-04:
Bummer to lose the functionality of TyCommander.

Two ways to 'Restore':
TyCommander has an option in the place where 'Integrate to Arduino' was done to 'Restore'
 ... That involves running TyCommander to complete that task. 

The second would be to Reinstall TeensyDuino that will overwrite the altered Boards.txt (IIRC) file.

Hack 3rd way would be to find that folder and see how TyCommander did the 'backup' - have been using TyCommander in alternative fashion to 'Integrate' so not sure if it is internal edit with comments as I guess - or if it is a file copy backup.

bdlabitt, 2022-08-04:
Reinstalled Arduino.  Reinstalled Teensyduino 1.56.  Can't program my Teensy4.1.  Get a blinking light not a continuous bright red LED during programming.  Installed Teensyduino 1.57.  No difference.  Cannot program my Teensy4.1.  But the Teensy is running and driving a display.  Not a happy camper.

Defragster, 2022-08-05:
If the forum post applies - that was due to a bad cable

bdlabitt, 2022-08-05:
Removed TyCommander, tycmd and tyuploader from /usr/local/bin.  Reinstalled Arduino.  Reinstalled Teensyduino 1.56.  Couldn't program any Teensy's or Arduinos.  Updated to Teensyduino 1.57.  Still couldn't program any Teensy's or Arduinos.  Replaced cable.  Now can program a Teensy and an AVR Arduino.

However, a hard crash still occurred.  Not a single interesting entry in the journal file.  The crash occurred 10 minutes after the last Teensy flashing!  I am now running PopOS kernel 5.18.10.  (Kernels prior to 5.15.0-37 seem to be ok.)  I did not use Teensy or Arduino for two weeks and had zero crashes.  I have had zero crashes prior to Teensyduino.

Does tytools use any of the Teensyduino or Teensy code?  Is it possible to use tycmd in a script (without Arduino) to be a more minimal example that one could debug with?  I have no idea how to get to closure.

Koromix, 2022-11-25:
Closing since this seems to be solved?

closed 2022-11-25 by Koromix